OpenCyphal-Garage/libcyphal #8: #If X==Y is true when both are not defined

davids5, 2015-01-08:
We ran into an issues with some of PX4 code where the construct !If X==Y is true when both are not defined. 

For example:

```
//#define PIGS 1
//#define DOGS 2

#if (DOGS == PIGS)
#warning("DOGS == PIGS") // it is to a compiler
#else
#warning "DOGS != PIGS"
#endif
```
When then built with -Wundef to ferret out any others:

warning: "UAVCAN_DEBUG" is not defined [-Wundef] UAVCAN_DEBUG
warning: "UAVCAN_MEM_POOL_BLOCK_SIZE" is not defined [-Wundef] UAVCAN_MEM_POOL_BLOCK_SIZE
warning: "UAVCAN_STM32_CHIBIOS" is not defined [-Wundef] UAVCAN_STM32_CHIBIOS
warning: "UAVCAN_STM32_TIMER_NUMBER" is not defined [-Wundef] UAVCAN_STM32_TIMER_NUMBER

@LorenzMeier asked me to 1) bring these to your attention to see if any of them are significant 
 and 2) see if the warnings can be remedied with an alternate construct of #if defined(XYZ) #if XYZ==PQT .... #endif #endif or such.

Thank you,

David

pavel-kirienko, 2015-01-08:
Thanks David.

These warnings are false positives as there is no real problem. The rationale for use of `#if` instead of `#ifdef` is provided [here](http://wiki.ros.org/CppStyleGuide#Preprocessor_directives_.28.23if_vs._.23ifdef.29).

I will provide a fix for these warnings tomorrow. Could you tell me what branch of PX4 Firmware should I use to reproduce the issue?

davids5, 2015-01-08:
Thank you! 

Good to know there are no real problems. 

Yes that rationale makes sense. 

https://github.com/PX4/Firmware/tree/nuttx_bringup but you will have to add the -Wundef to https://github.com/PX4/Firmware/blob/nuttx_bringup/makefiles/toolchain_gnu-arm-eabi.mk#L129

closed 2015-01-09 by pavel-kirienko