OpenCyphal-Garage/libcyphal #11: DSDL Actuator

Stapelzeiger, 2015-02-05:
The current actuator is for position control only. It might be useful to support other types (speed / torque controlled) actuators in the future. So I suggest renaming the actuator definition to indicate that they control position.

pavel-kirienko, 2015-02-05:
I guess we should extend the existing command message rather than defining a new one. The first solution that comes to mind is to add a field for command type:

- unitless -1..+1
- linear/angular position [meter or radian]
- linear/angular force [Newton or Newton metre]
- linear/angular speed [m/s or rad/s]

I'm not sure though if it would be useful to distinguish linear and angular commands.

Also, it seems like the Status message should have a speed field as well, and the `power` field should probably be dropped in order to keep the Status message just one CAN frame long.

Stapelzeiger, 2015-02-05:
This sounds good. I think it isn't necessary to explicitly distinguish the linear and angular type in the message since for a specific actuator it is evident whether it is linear or angular.

I agree about the speed field for the status. The power could then still be computed if the speed and torque/force is given.

pavel-kirienko, 2015-02-06:
@Stapelzeiger I just updated the definition. Please re-open if there's something missing.

commit 444937b8 (2015-02-06): Actuator messages update. Fixes #11
closed 2015-02-06 by pavel-kirienko