OpenCyphal-Garage/libcyphal #22: Tutorial not updated

miccio-dk, 2015-04-13:
hi! i don't know if that's the most appropriate place to signal this, but after your [latest commit](https://github.com/UAVCAN/uavcan/commit/0c8f0700af72baf1f43829620feccfa647a992a4) of [uavcan/dsdl/uavcan/protocol/param/Value.uavcan](https://github.com/UAVCAN/uavcan/blob/master/dsdl/uavcan/protocol/param/Value.uavcan) the tutorial at http://uavcan.org/Libuavcan_platform_documentation:_Linux needs to be updated by changing `ValueString` to `String`, as right now it wouldn't compile. I would do it myself but I can't edit the wiki, maybe somebody in here can. 
Keep up the good work!

pavel-kirienko, 2015-04-13:
Fixed, thanks Riccardo!

I will do another pass on the tutorials and all the code samples when the last RFC is released (soon).

Pavel.

closed 2015-04-13 by pavel-kirienko

miccio-dk, 2015-04-14:
Great, thanks for all the useful documentation!