OpenCyphal-Garage/libcyphal #28: Implement POSIX file server backend

pavel-kirienko, 2015-05-18:
@davids5 Can you help with it?

Here's the backend interface: https://github.com/UAVCAN/uavcan/blob/dynamic_node_id_raft/libuavcan/include/uavcan/protocol/file_server.hpp#L24. For the firmware update server we'll need at least `getInfo()` and `read()`.

The backend class should go into `libuavcan_drivers/posix/include/uavcan_posix/file_server_backend.hpp`.

https://github.com/UAVCAN/uavcan/pull/27 needs to be merged first.

davids5, 2015-05-20:
Hi Pavel,

 

This has been pushed.

 

David

pavel-kirienko, 2015-05-20:
Great, thank you!

A few questions:

1. You defined a new string type - [PathString](https://github.com/UAVCAN/uavcan/blob/dynamic_node_id_raft/libuavcan_drivers/posix/include/uavcan_posix/file_server_backend.hpp#L43), of max length (200+128) bytes. That's a lot. Since we're concerned about memory use, are you sure we need prefixes that long? Couldn't we keep it at 200 bytes long? This would allow to re-use the existing string type - `IFileServerBackend::Path` - for the prefix, which should also reduce code bloat because of fewer instantiations of the string template.

2. [Here](https://github.com/UAVCAN/uavcan/blob/dynamic_node_id_raft/libuavcan_drivers/posix/include/uavcan_posix/file_server_backend.hpp#L93) we should also set flags `FLAG_READABLE` and `FLAG_WRITEABLE`, until proper permission checks are implemented.

3. Return codes are different from the rest of the library; I failed to document it properly in the interface description (will fix that). See https://github.com/UAVCAN/uavcan/blob/master/dsdl/uavcan/protocol/file/Error.uavcan#L9

davids5, 2015-05-20:
[DBS] Yes it is a lot. The concern I had was the following. The server may have a "base path" to the file files it is willing to server. The protocol defines a path of 200. But that does not include the base path. Therefore any request where strlen(path+" base path") > 200 would fail or worse could truncate a file name to a meaningful name of a device but not it's hardware revision. Could be I am just over engineering it. What do you think?

[DBS] I punted on the flags. I need a better education on the expected use case.

[DBS] OK that makes sense. Fixing it now.

closed 2015-05-20 by pavel-kirienko