OpenCyphal-Garage/libcyphal #32: POSIX tools refactoring

pavel-kirienko, 2015-05-26:
@davids5 I reviewed the firmware file management logic in the POSIX tools module, and introduced some style fixes and comments. I believe the classes `FirmwareCommon` and `FirmwareTools` should be removed, and their logic should be moved inside `FirmwareVersionChecker`. Please see the TODO comments I placed in the code.

Thanks!

davids5, 2015-05-27:
@pavel-kirienko FirmwareCommon is gone.  By FirmwareTools if you mean the FirmarePath - I can remove that in the AM. Do you want the class it ends up in to absorb and abstracting the file path length issues by supporting the creations of the paths  or should the code that uses the library have to deal with it?

pavel-kirienko, 2015-05-27:
Thanks David!

> Do you want the class it ends up in to absorb and abstracting the file path length issues by supporting the creations of the paths or should the code that uses the library have to deal with it?

My suggestion is to move all of the firmware management logic into `FirmwareVersionChecker` or some namespace, in order to avoid pollution of the API, so it's the first option you described.

davids5, 2015-05-27:
Changes pushed. See if you like the macro for Nuttx/Linux

pavel-kirienko, 2015-05-28:
Thanks, it's okay.

closed 2015-05-29 by pavel-kirienko