OpenCyphal-Garage/libcyphal #33: Memory footprint is large

davids5, 2015-05-27:
Distributed

   text    data     bss     dec     hex filename
 892504    2712   35464  930680   e3378 Build/px4fmu-v2_default.build/firmware.elf

147 uavcan                       5545  3.282  2856/ 3380 240 (240)  w:sem

Centralized
  text    data     bss     dec     hex filename
 871560    2712   35352  909624   de138 Build/px4fmu-v2_default.build/firmware.elf

156 uavcan                        568  2.852  2856/ 3380 240 (240)  w:sem

Centralized save 20K  of FLASH but **ONLY 112 Bytes of RAM**?

Using the Node allocation and firmware distribution costs as compared to master

43K of flash 2480 Bytes SRAM and over Master @ 5c53d38652dc6c97e216ea6b70215a95890df572
text    data     bss     dec     hex filename
 827240    2948   32872  863060   d2b54 Build/px4fmu-v2_default.build/firmware.elf

davids5, 2015-05-28:
To compare apples to apples I added added #define UAVCAN_FW_SERVER

Without UAVCAN_FW_SERVER defined:
  text    data     bss     dec     hex filename
 **831496**    2708   34600  868804   d41c4 Build/px4fmu-v2_default.build/firmware.elf

             total       used       free    largest
Mem:        224352     168096      56256      **52288**
nsh> uavcan start
uavcan: Node ID 12, bitrate 1000000
uavcan: SW version vcs_commit: 0xacc4ded4
uavcan: sensor bridge 'gnss' init ok
uavcan: sensor bridge 'mag' init ok
uavcan: sensor bridge 'baro' init ok
nsh> free
             total       used       free    largest
Mem:        224352     194112      30240      **24832**

PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE
  151 uavcan                        249  0.710   **920**/ 3380 240 (240)  w:sem

-------------------------------------------------------------------------------------------------------------------------------
with UAVCAN_FW_SERVER defined
    text    data     bss     dec     hex filename
**871184**    2712   35352  909248   ddfc0 Build/px4fmu-v2_default.build/firmware.elf

nsh> free
             total       used       free    largest
Mem:        223600     168608      54992      **51536**
nsh> uavcan start
uavcan: Node ID 12, bitrate 1000000
uavcan: SW version vcs_commit: 0xacc4ded4
uavcan: sensor bridge 'gnss' init ok
uavcan: sensor bridge 'mag' init ok
uavcan: sensor bridge 'baro' init ok
nsh> free
             total       used       free    largest
Mem:        223600     209760      13840       **7600**

PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE
 151 uavcan                        582  0.949  **2800**/ 3380 240 (240)  w:sem
 
Total file differences
[image]

pavel-kirienko, 2015-05-28:
On my AMD64 machine (GCC 4.8.2, Linux 3.13) I get the following sizes:

```
distributed::Server:                    11088
-----------------------------------------------------------
distributed::RaftCore:                  5784
    distributed::PersistentState:       3120
        distributed::Log:               3096
    distributed::ClusterManager:        432
AllocationRequestManager:               512
NodeDiscoverer:                         4736

centralized::Server:                    7464
-----------------------------------------------------------
centralized::Storage:                   2168
AllocationRequestManager:               512
NodeDiscoverer:                         4736
```

In the tables above, nesting of objects is indicated with indents. Apparantly, `NodeDiscoverer` is the largest memory consumer (`RaftCore` doesn't count as it is only used in the Distributed server), this is caused by the fact that it [statically allocates 64 receiver objects](https://github.com/UAVCAN/uavcan/blob/dynamic_node_id_raft/libuavcan/include/uavcan/protocol/dynamic_node_id_server/node_discoverer.hpp#L97-L109), which allows it to operate on a bus with this many nodes without using any pool memory (the memory pool will be used if there's more nodes on the bus). I don't have a solution for this, but there's something else that can be optimized.

Currently, the library [statically allocates one `ReceivedDataStruct<T>` object in every object of classes `Subscriber<T>`, `ServiceClient<T>`, and `ServiceClient<T>`](https://github.com/UAVCAN/uavcan/blob/master/libuavcan/include/uavcan/node/generic_subscriber.hpp#L165-L171). Despite the fact that the objects are typically small (under half a kilobyte), the cumulative memory use can be significant (for example, a rough estimate for the objects in the table above shows that total memory wasted on such allocation is close to 1K). A possible solution is to define a common buffer somewhere in the library, much like [`MarshalBufferProvider`](https://github.com/UAVCAN/uavcan/blob/master/libuavcan/include/uavcan/node/marshal_buffer.hpp#L50), which then will be used by templates listed above whenever a transfer is ready to be demarshalled.

pavel-kirienko, 2015-05-28:
With the common buffer:

```
centralized::Server:          7024
distributed::Server:          10528
```

pavel-kirienko, 2015-05-28:
`sizeof(uavcan::Node<0>)` has reduced by about 700 bytes.

davids5, 2015-05-29:
Will -flto help? Not really but -Os on the uavcan module will:

nuttx_next_update_to_uavcan_raft_probe as built -O3 

```
text    data     bss     dec     hex filename
 871904    2708   35576  910188   de36c /home/david_s5/Desktop/dev/PX4/repos/mainline/Firmware/Build/px4fmu-v2_default.build/firmware.elf
```
- Os on uavcan module
```
846576    2708   35576  884860   d807c 
```
- Os on uavcan module adding -flto, and passing -Os to linker
```
 846272    2708   35576  884556   d7f4c
```

pavel-kirienko, 2015-05-29:
The shared buffer optimization I described above resulted in the following gains:

Before optimization (current HEAD):

```
$ cat libuavcan_test.log | grep sizeof
sizeof(uavcan::Node<0>): 5744
sizeof(uavcan::Publisher<uavcan::mavlink::Message>): 72
sizeof(uavcan::Subscriber<uavcan::mavlink::Message, Listener::ExtendedBinder>): 560
sizeof(uavcan::Subscriber<root_ns_a::EmptyMessage, Listener::SimpleBinder>): 272
sizeof(uavcan::NodeInfoRetriever): 6480
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 1264
sizeof(uavcan::NodeInfoRetriever): 6480
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 1264
sizeof(uavcan::BasicFileServer): 1952
sizeof(uavcan::FileServer): 5056
sizeof(uavcan::NetworkCompatibilityChecker): 1368
sizeof(uavcan::FirmwareUpdateTrigger): 920
```

After optimization:
```
$ cat log | grep sizeof
sizeof(uavcan::Node<0>): 5408
sizeof(uavcan::Publisher<uavcan::mavlink::Message>): 72
sizeof(uavcan::Subscriber<uavcan::mavlink::Message, Listener::ExtendedBinder>): 440
sizeof(uavcan::Subscriber<root_ns_a::EmptyMessage, Listener::SimpleBinder>): 264
sizeof(uavcan::NodeInfoRetriever): 6072
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 872
sizeof(uavcan::NodeInfoRetriever): 6072
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 872
sizeof(uavcan::BasicFileServer): 1520
sizeof(uavcan::FileServer): 3776
sizeof(uavcan::NetworkCompatibilityChecker): 1072
sizeof(uavcan::FirmwareUpdateTrigger): 776
```

pavel-kirienko, 2015-05-29:
Comparison for the LPC11C24 test app:

`dynamic_node_id_raft`:

```
$ arm-none-eabi-size build/firmware.elf 
   text    data     bss     dec     hex filename
  23588      44    5344   28976    7130 build/firmware.elf
```

`global_rx_object_buffer`

```
```

pavel-kirienko, 2015-05-29:
Another build of the LPC11C24, this time the argument `MarshalBufferSize` of `Node<>` is set to 264, same as in the `dynamic_node_id_raft` branch:

```
   text    data     bss     dec     hex filename
  23564      44    5320   28928    7100 build/firmware.elf
```

ROM reduction 24 bytes, RAM reduction 24 bytes.

pavel-kirienko, 2015-05-29:
LPC11C24, RX objects are stack-allocated, marshal buffer size is 376 bytes:

```
   text    data     bss     dec     hex filename
  23556      44    5432   29032    7168 build/firmware.elf
```

pavel-kirienko, 2015-05-29:
Same build of the HEAD of `global_rx_object_buffer` - service response structure is now stack allocated:

```
   text    data     bss     dec     hex filename
  23492      44    5044   28580    6fa4 build/firmware.elf
```

This is a rather significant improvement in memory use.

Object sizes:

```
$ cat libuavcan_test.log | grep sizeof
sizeof(uavcan::Node<0>): 4464
sizeof(uavcan::Publisher<uavcan::mavlink::Message>): 72
sizeof(uavcan::Subscriber<uavcan::mavlink::Message, Listener::ExtendedBinder>): 440
sizeof(uavcan::Subscriber<root_ns_a::EmptyMessage, Listener::SimpleBinder>): 264
sizeof(uavcan::NodeInfoRetriever): 6072
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 872
sizeof(uavcan::NodeInfoRetriever): 6072
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 872
sizeof(uavcan::BasicFileServer): 1240
sizeof(uavcan::FileServer): 3288
sizeof(uavcan::NetworkCompatibilityChecker): 1072
sizeof(uavcan::FirmwareUpdateTrigger): 776
```

For reference, this is what we had before optimization:

```
$ cat libuavcan_test.log | grep sizeof
sizeof(uavcan::Node<0>): 5744
sizeof(uavcan::Publisher<uavcan::mavlink::Message>): 72
sizeof(uavcan::Subscriber<uavcan::mavlink::Message, Listener::ExtendedBinder>): 560
sizeof(uavcan::Subscriber<root_ns_a::EmptyMessage, Listener::SimpleBinder>): 272
sizeof(uavcan::NodeInfoRetriever): 6480
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 1264
sizeof(uavcan::NodeInfoRetriever): 6480
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 1264
sizeof(uavcan::BasicFileServer): 1952
sizeof(uavcan::FileServer): 5056
sizeof(uavcan::NetworkCompatibilityChecker): 1368
sizeof(uavcan::FirmwareUpdateTrigger): 920
```

pavel-kirienko, 2015-05-29:
Deltas:
[image]

pavel-kirienko, 2015-05-29:
Latest optimization (that was pending for several months with a dedicated TODO comment):

```
   text    data     bss     dec     hex filename
  23388      44    4980   28412    6efc build/firmware.elf
```

```
$ cat libuavcan_test.log | grep sizeof
sizeof(uavcan::Node<0>): 4352
sizeof(uavcan::Publisher<uavcan::mavlink::Message>): 56
sizeof(uavcan::Subscriber<uavcan::mavlink::Message, Listener::ExtendedBinder>): 440
sizeof(uavcan::Subscriber<root_ns_a::EmptyMessage, Listener::SimpleBinder>): 264
sizeof(uavcan::NodeInfoRetriever): 6056
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 856
sizeof(uavcan::NodeInfoRetriever): 6056
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 856
sizeof(uavcan::BasicFileServer): 1208
sizeof(uavcan::FileServer): 3208
sizeof(uavcan::NetworkCompatibilityChecker): 1056
sizeof(uavcan::FirmwareUpdateTrigger): 760
```

davids5, 2015-05-30:
Good work @pavel-kirienko !!!!!

Using dynamic_node_id_raft@cf32ca057324c5423fc63b5fdcee2e6506c76cfc 

uavcan node doing ESC RPM etc protocol with LED Uses less RAM!!!!
```
   text    data     bss     dec     hex filename
  99484     416    3900  103800   19578 Build/px4cannode-v1_default.build/firmware.elf

UAVCAN  getNumFreeBlocks in bytes 1008
              total       used       free    largest
Data:         15920       9896       6024       **4216** Yeah - this is up from ~2500!
Prog:        132096     110592      21504      21504
   2 uavcan           2935  1584/ 1660 240 (240)  RUN
```

Boot loader is smaller by ~208 Bytes - that is FANTASTIC!!!
```
   text    data     bss     dec     hex filename
   **7971**       0    3552   11523    2d03 Build/px4cannode-v1_bootloader.build/firmware.elf
```

FMU code is still using a lot  of SRAM (~34K) to run uavcan. That is steep 
As predicted stack utilization went up 416 Bytes from 2800 to 3216
```
   text    data     bss     dec     hex filename
 870512    2708   35368  908588   ddd2c Build/px4fmu-v2_default.build/firmware.elf
             total       used       free    largest
Mem:        223584     168608      54976      51440
nsh> uavcan start
uavcan: Node ID 12, bitrate 1000000
uavcan: SW version vcs_commit: 0x04b2126e
uavcan: sensor bridge 'gnss' init ok
uavcan: sensor bridge 'mag' init ok
uavcan: sensor bridge 'baro' init ok
nsh> free
             total       used       free    largest
Mem:        223584     206944      16640      11536
```
```
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE
  151 uavcan                       3326  6.888  3216/ 3380 240 (240)  w:sem
 ```

pavel-kirienko, 2015-05-30:
We could cut another couple of hundred of bytes of both RAM and ROM by means of using stack-allocated marshaling buffer instead of a shared one. Stack use will go up a little though, but cumulative footprint should still be lower.

pavel-kirienko, 2015-05-31:
Shared marshaling buffer removed, now using stack allocation. Results:

LPC11C24:

```
   text    data     bss     dec     hex filename
  23224      44    4592   27860    6cd4 build/firmware.elf
```

Tests:

```
sizeof(uavcan::Node<0>): 3824
sizeof(uavcan::Publisher<uavcan::mavlink::Message>): 48
sizeof(uavcan::Subscriber<uavcan::mavlink::Message, Listener::ExtendedBinder>): 440
sizeof(uavcan::Subscriber<root_ns_a::EmptyMessage, Listener::SimpleBinder>): 264
sizeof(uavcan::NodeInfoRetriever): 6048
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 848
sizeof(uavcan::NodeInfoRetriever): 6048
sizeof(uavcan::ServiceClient<uavcan::protocol::GetNodeInfo>): 848
sizeof(uavcan::BasicFileServer): 1192
sizeof(uavcan::FileServer): 3160
sizeof(uavcan::NetworkCompatibilityChecker): 1048
sizeof(uavcan::FirmwareUpdateTrigger): 752
```

`Node<0>` reduced by 501 bytes, LPC11C24 RAM/flash use reduced by 388/164 bytes.

davids5, 2015-05-31:
@pavel-kirienko  How deep is the stack penetration on the LPC11C24?

pavel-kirienko, 2015-05-31:
About 1.5K.

davids5, 2015-05-31:
What UAVCAN specs are included in the build?

pavel-kirienko, 2015-05-31:
The node is publishing mandatory `uavcan.protocol.NodeStatus`, synchronizes its local clock with the network time using time synchronization slave (message `uavcan.protocol.GlobalTimeSync`), responds to `uavcan.protocol.GetNodeInfo`, and publishes some log messages via `uavcan.protocol.debug.LogMessage`.

The whole app is here: https://github.com/UAVCAN/uavcan/blob/master/libuavcan_drivers/lpc11c24/test_olimex_lpc_p11c24/src/main.cpp

davids5, 2015-05-31:
The  LPC11C24 part has 32 FLASH and 8 K RAM - correct ?

As the bootloader is now, it builds at 8K which leaves the total APP space is 24K. As built the the APP code above is taking 22.67K 

If we add a TINY to the bootloader, remove all LED, uavcan debug.log, It may be 6K leaving 26K for the App. This means we have   1.3 - 3.3 K to implement features.

That is super tight!

pavel-kirienko, 2015-05-31:
> The LPC11C24 part has 32 FLASH and 8 K RAM - correct ?

Yes, but remember that it is just a test application that runs some random logic only for testing purposes. The baseline is as follows:

```
   text    data     bss     dec     hex filename
  19616      44    4244   23904    5d60 build/firmware.elf
```

The above build only runs `uavcan.protocol.NodeStatus` and `uavcan.protocol.GetNodeInfo`, and blinks a LED.

So the baseline build takes 19K of flash, add 8K of bootloader (6K doesn't make sense because flash page size is 4K), you'll get 5K for the application, it's not that bad.

davids5, 2015-05-31:
What happens if you add the the subs/pubs for an ESC and an LED? What is left for the actual esc and led driver/application blocks?

pavel-kirienko, 2015-05-31:
I guess subs/pubs should take about 1K, will check tomorrow.

pavel-kirienko, 2015-06-01:
Added subs/pubs to the LPC11C24 test, here's the full code:

```c++
/*
 * Copyright (C) 2014 Pavel Kirienko <pavel.kirienko@gmail.com>
 */

#include <cstdio>
#include <algorithm>
#include <board.hpp>
#include <chip.h>
#include <type_traits>
#include <uavcan_lpc11c24/uavcan_lpc11c24.hpp>
#include <uavcan/equipment/esc/RawCommand.hpp>
#include <uavcan/equipment/esc/Status.hpp>
#include <uavcan/equipment/indication/LightsCommand.hpp>

namespace
{

typedef uavcan::Node<2800> Node;

Node& getNode()
{
    static Node node(uavcan_lpc11c24::CanDriver::instance(), uavcan_lpc11c24::SystemClock::instance());
    return node;
}

template <typename Msg>
uavcan::Subscriber<Msg, void (*)(const uavcan::ReceivedDataStructure<Msg>&)>&
getSubscriber()
{
    static typename std::remove_reference<decltype(getSubscriber<Msg>())>::type obj(getNode());
    return obj;
}

template <typename Msg>
uavcan::Publisher<Msg>& getPublisher()
{
    static uavcan::Publisher<Msg> obj(getNode());
    return obj;
}

void handleEscRawCommand(const uavcan::ReceivedDataStructure<uavcan::equipment::esc::RawCommand>&)
{
    ; // Doing nothing
}

void handleLightsCommand(const uavcan::ReceivedDataStructure<uavcan::equipment::indication::LightsCommand>&)
{
    ; // Doing nothing
}

#if __GNUC__
__attribute__((noreturn))
#endif
void die()
{
    while (true) { }
}

#if __GNUC__
__attribute__((noinline))
#endif
void init()
{
    board::resetWatchdog();

    if (uavcan_lpc11c24::CanDriver::instance().init(1000000) < 0)
    {
        die();
    }

    board::resetWatchdog();

    getNode().setNodeID(72);
    getNode().setName("org.uavcan.lpc11c24_test");

    uavcan::protocol::SoftwareVersion swver;
    swver.major = FW_VERSION_MAJOR;
    swver.minor = FW_VERSION_MINOR;
    swver.vcs_commit = GIT_HASH;
    swver.optional_field_mask = swver.OPTIONAL_FIELD_MASK_VCS_COMMIT;
    getNode().setSoftwareVersion(swver);

    uavcan::protocol::HardwareVersion hwver;
    std::uint8_t uid[board::UniqueIDSize] = {};
    board::readUniqueID(uid);
    std::copy(std::begin(uid), std::end(uid), std::begin(hwver.unique_id));
    getNode().setHardwareVersion(hwver);

    board::resetWatchdog();

    while (getNode().start() < 0)
    {
    }

    board::resetWatchdog();

    while (getSubscriber<uavcan::equipment::esc::RawCommand>().start(handleEscRawCommand) < 0)
    {
    }

    while (getSubscriber<uavcan::equipment::indication::LightsCommand>().start(handleLightsCommand) < 0)
    {
    }

    board::resetWatchdog();
}

}

int main()
{
    init();

    getNode().setStatusOk();

    uavcan::MonotonicTime prev_log_at;

    while (true)
    {
        const int res = getNode().spin(uavcan::MonotonicDuration::fromMSec(25));
        board::setErrorLed(res < 0);
        board::setStatusLed(uavcan_lpc11c24::CanDriver::instance().hadActivity());

        const auto ts = uavcan_lpc11c24::clock::getMonotonic();
        if ((ts - prev_log_at).toMSec() >= 1000)
        {
            prev_log_at = ts;

            (void)getPublisher<uavcan::equipment::esc::Status>().broadcast(uavcan::equipment::esc::Status());
        }

        board::resetWatchdog();
    }
}
```

Results:

```
   text    data     bss     dec     hex filename
  27242     144    4536   31922    7cb2 build/firmware.elf
```

davids5, 2015-06-01:
Latest figures:

FMU:
Ram sizes looking better.

Odd - removed code yet code size when up. 

```
LAST:
   text    data     bss     dec     hex filename
 870512    2708   35368  908588   ddd2c Build/px4fmu-v2_default.build/firmware.elf
             total       used       free    largest
Mem:        223584     168608      54976      51440
nsh> uavcan start
uavcan: Node ID 12, bitrate 1000000
uavcan: SW version vcs_commit: 0x04b2126e
uavcan: sensor bridge 'gnss' init ok
uavcan: sensor bridge 'mag' init ok
uavcan: sensor bridge 'baro' init ok
nsh> free
             total       used       free    largest
Mem:        223584     206944      16640      11536
 151 uavcan                       5746  0.473  3216/ 3380 240 (240)  w:sem

Current

870880    2708   35368  908956   dde9c Build/px4fmu-v2_default.build/firmware.elf
Mem:        223584     168016      55568      51472
nsh> uavcan start
uavcan: Node ID 12, bitrate 1000000
uavcan: SW version vcs_commit: 0x776e46a1
uavcan: sensor bridge 'gnss' init ok
uavcan: sensor bridge 'mag' init ok
uavcan: sensor bridge 'baro' init ok
nsh> free
             total       used       free    largest
Mem:        223584     205872      17712      12464
```

Node Stack crashes - But manageable  
Had to increase stack from 1660 to 2400 to get it to run 
Stack penetration went from 1584 to 2004 but because the static from 2696 (corrected) to 4008

when I set the Stack at 2100 (with 96 bytes headroom)

This is an overall improvement of 1192 bytes!!!!!

```
LAST:

  text    data     bss     dec     hex filename
  99484     416    3900  103800   19578 Build/px4cannode-v1_default.build/firmware.elf

UAVCAN  getNumFreeBlocks in bytes 1008
              total       used       free    largest
Data:         15920       9896       6024       **4216** Yeah - this is up from ~2500!
Prog:        132096     110592      21504      21504
   2 uavcan           2935  1584/ 1660 240 (240)  RUN

Current:
99388     416    3884  103688   19508 Build/px4cannode-v1_default.build/firmware.elf
              total       used       free    largest
Data:         15936      10120       5816       4008
Prog:        132096     110592      21504      21504
   2 uavcan            870  **2004**/ 2380 240 (240)  RUN

```

No Change:
```
7971       0    3552   11523    2d03 Build/px4cannode-v1_bootloader.build/firmware.elf
```

pavel-kirienko, 2015-06-01:
Can we tolerate these results, or do we need to optimize further? Continuing optimizations can be very time consuming, as at the moment I have no idea what else is left to optimize.

If we chose to continue, it would help if you could move all objects from heap to `.bss` and then post a linker map file, so we could see object sizes. In fact, such a map file would be nice to have even if we decided to give up on further optimizations.

pavel-kirienko, 2015-06-01:
Memory consumption issue was also mentioned here: https://github.com/PX4/Firmware/issues/2081.

I wonder where @LorenzMeier would put the current results on a scale of one to unacceptable.

davids5, 2015-06-01:
Update the cost of File Server and Node Allocation 

The Negative numbers reflect the additional resources used when UAVCAN_FW_SERVER is defined.

[image]

[image]

In my opinion, we need to have a long term plan for the memory diet. As we discussed there may be some other things we can do short of a C version of the library.  @LorenzMeier  needs to ultimately make the call. 

Static allocation can be done, but when uavcan is not running it will burn a lot of resources and I am not sure that would be a good thing to commit. If you just want a map file we can manage that if @LorenzMeier wants this worked on more.

pavel-kirienko, 2015-06-03:
Deferring until we get in a bind with memory.

davids5, 2015-06-12:
@pavel-kirienko, @LorenzMeier   Just an FYI Latest master 

px4cannode Moving in a good direction 96+348 more BYTES sram free
```
  text    data     bss     dec     hex filename
   7883       0    3552   11435    2cab Build/px4cannode-v1_bootloader.build/firmware.elf

UAVCAN  getNumFreeBlocks in bytes 1008
              total       used       free    largest
Data:         15936       9816       6120       **4312** Save another 100 bytes
Prog:        132096     110592      21504      21504
   2 uavcan           5818  1928/ 2084 240 (240)  RUN
```

px4fmu - Not So good using 17,104 bytes more 

```

  text    data     bss     dec     hex filename
 896344    2692   35276  934312   e41a8 Build/px4fmu-v2_default.build/firmware.elf

nsh> free
             total       used       free    largest
Mem:        223680     139904      83776      82320
nsh> mavlink_if0: Disabling hardware flow control
auvcan
nsh: auvcan: command not found
nsh> uavcan start
uavcan: Node ID 1, bitrate 1000000
uavcan: SW version vcs_commit: 0xcab320d9
uavcan: sensor bridge 'gnss' init ok
uavcan: sensor bridge 'mag' init ok
uavcan: sensor bridge 'baro' init ok
nsh> free
             total       used       free    largest
Mem:        223680     177600      46080      39952
```

[image]

pavel-kirienko, 2015-06-13:
17k! This is insane. Any chance to bisect to the exact commit which added the extra memory?

davids5, 2015-06-13:
[image]

davids5, 2015-06-13:
[image]

davids5, 2015-06-13:
[image]

davids5, 2015-06-13:
@pavel-kirienko the 17K vrs 13K may have been nuttx_next to master, but above are the numbers.
px4 uavcan for esc, etc SRAM costs 38,752 bytes with and 24,880 bytes without the FW/Allocation Server

pavel-kirienko, 2015-06-13:
Thanks David.

We can shed maybe a hundred bytes of RAM and ROM by optimizing the block allocator. This optimization will also simplify the code and it will not cause any usability penalties, so it's a win-win.

Another way of optimization is to make these two constants compile-time configurable:
- https://github.com/UAVCAN/libuavcan/blob/master/libuavcan/include/uavcan/protocol/dynamic_node_id_server/node_discoverer.hpp#L97
- https://github.com/UAVCAN/libuavcan/blob/master/libuavcan/include/uavcan/protocol/node_status_monitor.hpp#L57

If the number of nodes on the bus exceeds the value of this constant, the library will take extra memory from the pool (one block per extra node).

Third optimization is to make the dynamic node ID allocation server optional, so that those applications that don't care about it may keep it disabled while still using the rest of the UAVCAN driver functionality. This optimization is trivial to implement, and it will be even more trivial once the multithreaded node is implemented.

pavel-kirienko, 2015-06-13:
Oh, an important comment concerning the second optimization: reducing the number of nodes from 64 (current value) to 32 should release about 2~3K.

pavel-kirienko, 2015-06-15:
@davids5 Please add `-DUAVCAN_MAX_NETWORK_SIZE_HINT=16` [here](https://github.com/PX4/Firmware/blob/master/src/modules/uavcan/module.mk#L68).

This should free a few K of memory.

davids5, 2015-06-15:
I am not sure a "few" K will help

@pavel-kirienko please see https://github.com/PX4/Firmware/issues/2381

pavel-kirienko, 2015-06-15:
I'm not implying that this issue will be fully resolved once these "few" K are freed. We have actually discussed this optimization earlier (in a call) and I said then that I'll implement it someday - well, here it is.

davids5, 2015-06-15:
@pavel-kirienko Sorry - emails crossed. Also there is a lot of conversation here. Would you think it would help to update the first post with three groups 
1) Done and saved,
2) Thing done that did not work 
3) Things left to do - as a check list

and link the text in the lists to the posts herein?

pavel-kirienko, 2015-06-16:
# Implemented optimizations

### Concurrent service calls

The class `uavcan::ServiceClient<>` has been refactored so that now it is possible to perform concurrent calls to multiple servers using one Client object.

This optimization was not recorded in the issues (was discussed in a series of voice calls). The RAM/ROM gains are not recorder either (around 3K IIRC).

### UAVCAN_MAX_NETWORK_SIZE_HINT

Number of pre-allocated receiver objects for subscriptions to `uavcan.protocol.NodeStatus` has been made configurable. This allowed to reduce memory use at the cost of requiring more dynamic memory if the node is operating in a large network.

RAM gains are unknown - waiting to hear from @davids5. Expectations are 2~3K.

Mentioned:
- https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-106261072
- https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-111724267

### Allocator refactoring

A useless class `uavcan::PoolManager` was removed, which saved 80 bytes of ROM on LPC11C24.

Mentioned: https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-111724267

### Stack allocated RX objects and internal adjustments

This optimization includes two internal changes:

- RX objects are allocated on the stack instead of using static allocation.
- Some internal fields were moved around a bit to avoid data duplication.

Mentioned:
- https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-106261072
- https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-106926210

Results are summarized here: https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-107074277

### Stack allocated TX marshaling buffers

Using stack allocation instead of a shared buffer.

Mentioned: https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-107075254

LPC11C24 RAM/flash use reduced by 388/164 bytes.

Overall improvement in the PX4 build is 1192 bytes (see [this comment](https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-107717287)).

# Vague ideas for future optimizations

### Do not cache the allocation table in RAM

Currently, the centralized allocation server keeps its entire allocation table in RAM as a write-through cache. This allows it to avoid requesting storage backend IO when it serves allocation requests for existing entries (which is a major use case).

According to [this](https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-106261072), the cache takes about 2K of RAM.

Disadvantage is that the server will be doing a lot of storage searches while serving allocation requests, i.e. after boot. @davids5 reports that FS search is a very time consuming operation (hundreds of milliseconds IIRC).

### Totally refactor the implementation of the transport layer

The current implementation is somewhat inefficient in terms of memory use, and it will take a man month to rewrite it from scratch and test. No idea how much benefit will it bring (likely no more than a couple of K).

Some of the ideas proposed in [this design document](https://docs.google.com/document/d/1mmL0MJ7gF3yogZPsXIpDp-cw-WDDLz7EyO7BoTem_wI/edit) could be applied in libuavcan as well.

### Reduce memory pool

It might be possible to reduce the memory pool block size to 48 bytes without breaking things much.

Also, the minimal required number of memory blocks should be carefully re-evaluated - maybe the current number of blocks (192) is too much.

pavel-kirienko, 2015-06-18:
Allocation table cache removed in master, 2144 bytes of memory freed at the expense of one more storage read operation per allocation request.

pavel-kirienko, 2015-06-18:
The latest master allows to reduce pool block size to 48 bytes on 32-bit machines, although I didn't have a chance to run it on a Cortex device yet.

Recipe:

```
-DUAVCAN_MEM_POOL_BLOCK_SIZE=48
-DUAVCAN_PACKED_BEGIN='_Pragma("pack(1)")' 
-DUAVCAN_PACKED_END='_Pragma("pack()")'
```

pavel-kirienko, 2015-06-20:
Memory use in PX4 is down to 14.5K, closing. A bit more memory can be reclaimed in PX4 by means of moving the uavcan objects from heap into bss.

davids5, 2015-07-01:
@pavel-kirienko @LorenzMeier 

Just documenting the new Frame format stats:

```
  text    data     bss     dec     hex filename
   7027       1    3552   10580    2954 Build/px4cannode-v1_bootloader.build/firmware.elf
   text    data     bss     dec     hex filename
 100632     416    3884  104932   199e4 Build/px4cannode-v1_default.build/firmware.elf
   text    data     bss     dec     hex filename
 908680    2696   34796  946172   e6ffc Build/px4fmu-v2_default.build/firmware.elf
   text    data     bss     dec     hex filename
  47229     516    3040   50785    c661 Build/px4io-v2_default.build/firmware.elf
   text    data     bss     dec     hex filename
   6631       1    3636   10268    281c Build/s2740vc-v1_bootloader.build/firmware.elf
   text    data     bss     dec     hex filename
  29419     320    2500   32239    7def Build/s2740vc-v1_default.build/firmware.elf
```

davids5, 2015-08-08:
Added UAVCAN Time Synchronization (Starts with uavcan start)

```
 text    data     bss     dec     hex filename
   8159       1    3580   11740    2ddc Build/px4cannode-v1_bootloader.build/firmware.elf
   text    data     bss     dec     hex filename
 100235     416    3980  104631   198b7 Build/px4cannode-v1_default.build/firmware.elf
   text    data     bss     dec     hex filename
 912792    2696   34892  950380   e806c Build/px4fmu-v2_default.build/firmware.elf
   text    data     bss     dec     hex filename
  47221     516    3056   50793    c669 Build/px4io-v2_default.build/firmware.elf

```
```
 free
             total       used       free    largest
Mem:        224064     165104      58960      56512
nsh> uavcan start
...
             total       used       free    largest
Mem:        224064     180496      43568      38496
```

pavel-kirienko, 2015-08-23:
This issue is unsinkable.

davids5, 2015-09-23:
latest Master + https://github.com/PX4/Firmware/pull/2895

```
Build datetime: Sep 23 2015 04:08:13
FW git-hash: 1996d2b55a71e1a713c7388cbca9cb6b67e97506
Toolchain: 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]
```

```
nsh> free
             total       used       free    largest
Mem:        231696     167264      64432      60432
nsh> uavcan start
<snip>
nsh> free
             total       used       free    largest
Mem:        231696     181840      49856      44336
nsh> uavcan start fw
nsh> free
             total       used       free    largest
Mem:        231696     200144      31552      23344
nsh> uavcan stop fw
nsh> free
             total       used       free    largest
Mem:        231696     181840      49856      44336
```

```
  text	   data	    bss	    dec	    hex	filename
   text	   data	    bss	    dec	    hex	filename
 739336	   1904	   9864	 751104	  b7600	Build/px4fmu-v1_default.build/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 930472	   2932	  26568	 959972	  ea5e4	Build/px4fmu-v2_default.build/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 761492	   2876	  23496	 787864	  c0598	Build/px4fmu-v2_multiplatform.build/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 533912	   2904	  18392	 555208	  878c8	Build/px4fmu-v2_test.build/firmware.elf
   text	   data	    bss	    dec	    hex	filename
```

PR #73 (2015-10-17): Footprint reduction
Memory management has been refactored to better suit the needs of feature-rich applications, such as UAVCAN driver in PX4.

As a side effect of this change, small applications also show some marginal improvement, e.g. LPC11C24 demo shows ROM reduction of 844 bytes and RAM reduction of 12 bytes.

The library now implements a shared pool allocator, more on this later.

Temporarily resolves #33.
closed 2015-10-17 by pavel-kirienko