OpenCyphal-Garage/libcyphal #35: Device Driver and IO manager needs edge triggered and abstraction

davids5, 2015-05-30:
To prevent 100% load in application using poll the device driver need to be re-worked to only raise events on poll for  HW TX Empty And TX queue Not Empty for ACTIVE interfaces.

Since the UAVCAN io and multiplexing on the interfaces is handled internally by the io layer and the only interface the application has is to give UAVCAN cpu is spin_xxx there a violation of the encapsulation and abstraction needed by having the notion of interface choice and state managed by the application. There should be only one event used on the FD and it should only be set buy uavcan when the i/o layer can send or receive data.

pavel-kirienko, 2015-06-01:
See the referenced commit. All events are edge-triggered now and treated as POLLIN, so the issue of overlapping events we discussed last Friday is now resolved.

The abstraction issue is also sort of resolved (with caveats): since all events are now edge-triggered, a situation with events being stuck permanently in active state is not possible anymore. Hence, the poll() configuration code that violates abstraction can now be removed from the application. The caveat is that such handling can sometimes cause the library to handle irrelevant events, e.g. the driver will wake up poll() on a TX event, but the library doesn't have anything to transmit.

pavel-kirienko, 2015-06-01:
Waiting for David to test it.

davids5, 2015-06-01:
I updated the app code to not check the tx queue and revered the poll to just POLLIN. Tested Working

commit 22787651 (2015-06-01): STM32 NuttX driver: Edge-triggered poll(), sort of fixes #35
commit 5442c0ac (2015-06-01): Revert "STM32 NuttX driver: Edge-triggered poll(), sort of fixes #35"

This reverts commit 22787651e699c58b18258a8cb01249b0f5f329cb.
closed 2015-06-01 by davids5