OpenCyphal-Garage/libcyphal #40: Proper file IO loops

pavel-kirienko, 2015-06-10:
File read/write should be performed in a loop. See:

* https://github.com/UAVCAN/uavcan/blob/master/libuavcan_drivers/posix/include/uavcan_posix/dynamic_node_id_server/file_event_tracer.hpp#L57
* https://github.com/UAVCAN/uavcan/blob/master/libuavcan_drivers/posix/include/uavcan_posix/dynamic_node_id_server/file_storage_backend.hpp#L84
* https://github.com/UAVCAN/uavcan/blob/master/libuavcan_drivers/posix/include/uavcan_posix/dynamic_node_id_server/file_storage_backend.hpp#L58
* https://github.com/UAVCAN/uavcan/blob/master/libuavcan_drivers/posix/include/uavcan_posix/firmware_version_checker.hpp#L128
* https://github.com/UAVCAN/uavcan/blob/master/libuavcan_drivers/posix/include/uavcan_posix/basic_file_server_backend.hpp#L405

davids5, 2015-08-19:
@pavel-kirienko see https://github.com/UAVCAN/libuavcan/pull/59 
please test on Linix

pavel-kirienko, 2015-08-20:
Thanks David! Merged.

closed 2015-08-20 by pavel-kirienko