OpenCyphal-Garage/libcyphal #42: FirmwareUpdateTrigger.MultiNode TEST fail

ilia-sheremet, 2015-06-19:
Noncyclic error during tests. Usually appears after first build and following testing, but sometimes may not. Usually error comes up during only libuavcan_test_optim, but sometimes during others.

Last time also was combined with another Fail test - dynamic_node_id_server_NodeDiscoverer.RestartAndMaxAttempts, during libuavcan_test_cpp03.

Here are the log files after first build, where all the tests has failed - https://drive.google.com/open?id=0B9wE0Um8rdPgfmdUR1llZ2ZLMm9TVEw1ejE3eF9nRXJVZUJJQjVVWGlyWExkbUwyQUppYUU&authuser=0

ilia-sheremet, 2015-06-19:
Processor - Intel Core i7-3517U @ 2.40 GHz

closed 2015-07-26 by pavel-kirienko

pavel-kirienko, 2015-07-26:
The offending test has been fixed.

For extra paranoia I suggest you (and everyone whose builds are failing because tests cannot run in real time) to disable parallel build, or at least limit the number of jobs to the number of physical cores (2 in your case): `make -j2`.

commit b09509c4 (2015-07-26): Test fix #42