OpenCyphal-Garage/libcyphal #46: Documentation out of sync with library

porcoesphino, 2015-06-30:
This commit:

https://github.com/UAVCAN/libuavcan/commit/2843c0a35e16e24be00eddee7cbf525308025166#diff-35b7a237998b2054ec6ba90c34a0490aL74

Changes the CAN ID field so that it no longer matches the documentation:

http://uavcan.org/CAN_bus_transport_layer_specification#CAN_ID_field

This image needs to be remade since it's easy to skim past text and would confuse people.

This has also made its way into the PX4 codebase:

- Change was made on 2015-04-30 08:26:25 +0300
- Current version in PX4/Firmware is 988e404586900754e7e1748c87c0ec12b757cf87 made on 2015-06-07 00:07:49 +0300

pavel-kirienko, 2015-06-30:
Thanks, this is a known issue, a big documentation update is coming.

porcoesphino, 2015-07-21:
Great :) It seems like the CAN IDs (including MSG IDs) have jumped around a bit recently. When can we expect them to be stable?

LorenzMeier, 2015-07-21:
Pretty much now. We're phasing in the last change set before a freeze of the lower-level protocol aspects.

pavel-kirienko, 2015-08-31:
http://uavcan.org/Specification/4._CAN_bus_transport_layer.html#id-field

closed 2015-08-31 by pavel-kirienko