OpenCyphal-Garage/libcyphal #51: Update the LPC11C24 driver

pavel-kirienko, 2015-07-05:
Missing features:

1. CAN acceptance filters.
2. Transmission aborts.
3. Proper sample point configuration.

pavel-kirienko, 2015-07-05:
@ilia-sheremet I know that you've got warm feelings about LPC11C24. Let me know if you're interested in this.

miccio-dk, 2015-07-05:
@pavel-kirienko I could be interested too. I'm basically done with porting the drivers to the LPC1768 (will send a pull request for your evaluation after it's all sorted and neat) and I might be able to spare some time improving the support for the LPC11Cx.

ilia-sheremet, 2015-07-06:
@pavel-kirienko At the moment I don't have any spare LPC boards with me, so I won't be able to work on that.

pavel-kirienko, 2015-07-09:
@miccio-dk Hi Riccardo, sorry for taking long to get back. I have merged a WIP into master today, and I don't expect any further changes to the protocol that could affect the drivers, so it is now safe to start working on the LPC driver.

## CAN acceptance filters

There's a stub already in place, it just has to be filled with code: https://github.com/UAVCAN/libuavcan/blob/master/libuavcan_drivers/lpc11c24/driver/src/can.cpp#L329-L335

## Transmission aborts

This feature has not been documented anywhere (except in our team-internal emails), but it will be documented with the big documentation update that should happen soon (it's one of the next things on my TODO list). In short: UAVCAN got a strategically important feature of dynamic node ID allocation. Due to restrictions of the CAN bus itself, its native CSMA/CD logic breaks during dynamic node ID allocation (it's interesting that UAVCAN is not alone here, e.g. J1939 suffers from the same issue, and CANopen applies a peculiar workaround that was deemed inappropriate for UAVCAN, but that's a different story). The problem manifests itself when multiple nodes are trying to transmit the same CAN ID with different CAN payload values at the same time, which triggers a bus error condition. Upon detection of this condition, the transmitting nodes will try to retransmit immediately, causing a collision again, effectively driving the bus into a deadlock state. The solution is to transmit those frames in a special mode where automatic retransmission on error is disabled; beware though that it's not the same as time-triggered communication mode, the difference is in that we still want the frames to be retransmitted after an arbitration failure, which won't be the case in the TT communication mode. Let's get to the issue already: libuavcan now supports a special IO flag known as [`CanIOFlagAbortOnError`](https://github.com/UAVCAN/libuavcan/blob/master/libuavcan/include/uavcan/driver/can.hpp#L125-L132). The library will set this flag for every outgoing frame that must not be retransmitted in case of a bus error. The driver must respect this flag by means of cancelling the transmission of the current frame if this flag set active. This is the error handler where this logic should be implemented: https://github.com/UAVCAN/libuavcan/blob/master/libuavcan_drivers/lpc11c24/driver/src/can.cpp#L406-L413.

For reference, this is how the same feature is implemented in the STM32 driver (it is already validated and tested): https://github.com/UAVCAN/libuavcan/blob/master/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp#L575-L585

When this feature is implemented, you may want to test it; here's a simple helper script that allows you to flood the bus with a given CAN frame:

```python
#!/usr/bin/env python

import pycanbus # https://bitbucket.org/pavel_kirienko/pycanbus
import sys

CAN_ID = 0x0
CAN_PAYLOAD = '\x00\x00\x00\x00\x00\x00\x00'

bus = pycanbus.Socket('can0')

succeeded = 0
failed = 0

while True:
    tail = chr(0b11000000 + (succeeded & 0b11111))
    try:
        bus.send(pycanbus.Frame(CAN_PAYLOAD + tail, CAN_ID, True))
        succeeded += 1
    except Exception, ex:
        failed += 1
    sys.stdout.write('%d/%d\r ' % (succeeded, failed))
    sys.stdout.flush()
```

## Proper sample point configuration

See http://uavcan.org/Hardware_design_recommendations#CAN_bus_parameters.

The sample point should be located as close as possible to 87.5% of the bit time. The code that requires change is this routine that calculates the required CAN controller timing settings for the given bitrate: https://github.com/UAVCAN/libuavcan/blob/master/libuavcan_drivers/lpc11c24/driver/src/can.cpp#L132-L159.

---

Are you interested to help out on this? Would be much appreciated!

pavel-kirienko, 2015-07-09:
Oh, also it would be nice if the driver was updated to the C++11 standard; it will make the code cleaner and safer.

miccio-dk, 2015-07-10:
@pavel-kirienko  thanks for the very detailed answer. I will start to look into that from next week.

pavel-kirienko, 2015-07-15:
@miccio-dk So, are you going to be available to work on this?

pavel-kirienko, 2015-07-20:
@miccio-dk nudge nudge.

miccio-dk, 2015-07-20:
Hi @pavel-kirienko ,
sorry for such a late reply, but job kept me busy for the last week. As I mention above, I would really love to contribute to the LPC11C24 drivers, and I will send you an email soon with more details. I will try to make it fit into this week: worst case scenario, next weekend is gonna rain here.. :)

pavel-kirienko, 2015-07-26:
Apropos sample point location, the algorithm that computes proper configuration of the bit timing register can be found here: https://github.com/UAVCAN/libuavcan/blob/master/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp#L181-L316

The linked code is specific to STM bxCAN, but it can be easily adapted for Bosch C_CAN that is used in LPC controllers (particularly in LPC11C24).

This page can be used as a reference: http://www.bittiming.can-wiki.info/

PR #71 (2015-10-13): Missing features for LPC11C24
* CAN acceptance filters
* Support for dynamic node ID allocation (transmission aborts)
* Bus-off recovery
* Proper CAN sample point configuration
* Support for old C++ standards has been dropped, now it requires C++11 or newer
* Added compiler version checks; the driver requires GCC 4.9 since GCC 4.8 is incapable or building this firmware correctly (because of the above, apparently)

The demo application now includes:

* Automatic CAN bus bit rate configuration
* Dynamic node ID allocation
* UART console

Other features were left in place.

This PR resolves #51.
closed 2015-10-13 by pavel-kirienko