OpenCyphal-Garage/libcyphal #55: Clang build error on OS X

bendyer, 2015-07-28:
Build of b09509c40781be0235ca7b31b75fedabcb9a3d17 fails on OS X 10.10.4 with the version of clang/LLVM bundled with Xcode 6.4:

```
MacBook-Pro:build bendyer$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.4.0
Thread model: posix
```

Error is:
```
MacBook-Pro:build bendyer$ make
Running from the source directory
[  0%] Built target libuavcan_dsdlc
[  3%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/protocol/uc_node_status_provider.cpp.o
In file included from /Users/bendyer/Projects/ARM/workspace/libuavcan/libuavcan/src/protocol/uc_node_status_provider.cpp:5:
In file included from /Users/bendyer/Projects/ARM/workspace/libuavcan/libuavcan/include/uavcan/protocol/node_status_provider.hpp:15:
In file included from /Users/bendyer/Projects/ARM/workspace/libuavcan/libuavcan/include/dsdlc_generated/uavcan/protocol/GetNodeInfo.hpp:16:
/Users/bendyer/Projects/ARM/workspace/libuavcan/libuavcan/include/dsdlc_generated/uavcan/protocol/HardwareVersion.hpp:159:19: error: use of overloaded operator '==' is ambiguous (with operand types
      'const typename ::uavcan::StorageType<typename FieldTypes::unique_id>::Type' (aka 'const uavcan::Array<uavcan::IntegerSpec<8, 0, 0>, 0, 16>') and 'const typename ::uavcan::StorageType<typename FieldTypes::unique_id>::Type')
        unique_id == rhs.unique_id &&
        ~~~~~~~~~ ^  ~~~~~~~~~~~~~
/Users/bendyer/Projects/ARM/workspace/libuavcan/libuavcan/src/protocol/uc_node_status_provider.cpp:145:37: note: in instantiation of member function 'uavcan::protocol::HardwareVersion_<0>::operator==' requested here
    if (node_info_.hardware_version == protocol::HardwareVersion())
                                    ^
/Users/bendyer/Projects/ARM/workspace/libuavcan/libuavcan/include/uavcan/marshal/array.hpp:764:5: note: candidate function [with R = uavcan::Array<uavcan::IntegerSpec<8, 0, 0>, 0, 16>]
    operator==(const R& rhs) const
    ^
/Users/bendyer/Projects/ARM/workspace/libuavcan/libuavcan/include/uavcan/marshal/array.hpp:1090:13: note: candidate function [with R = uavcan::Array<uavcan::IntegerSpec<8, 0, 0>, 0, 16>, T = uavcan::IntegerSpec<8, 0, 0>, ArrayMode = 0, MaxSize = 16]
inline bool operator==(const R& rhs, const Array<T, ArrayMode, MaxSize>& lhs)
            ^
1 error generated.
make[2]: *** [libuavcan/CMakeFiles/uavcan.dir/src/protocol/uc_node_status_provider.cpp.o] Error 1
make[1]: *** [libuavcan/CMakeFiles/uavcan.dir/all] Error 2
make: *** [all] Error 2
```

The same version builds fine on Ubuntu with GCC 4.9.1:

```
/mnt/hgfs/workspace/libuavcan/build$ gcc --version
gcc (Ubuntu 4.9.1-16ubuntu6) 4.9.1
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

pavel-kirienko, 2015-08-15:
@bendyer would you mind testing https://github.com/UAVCAN/libuavcan/commit/e32dfafbae37ff995f165f7195eb847eaf60254d

bendyer, 2015-08-15:
That fixes it:

```
MacBook-Pro:build bendyer$ git checkout e32dfafbae37ff995f165f7195eb847eaf60254d
M	dsdl
Note: checking out 'e32dfafbae37ff995f165f7195eb847eaf60254d'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at e32dfaf... First stab at fixing #55
MacBook-Pro:build bendyer$ make
Running from the source directory
[  0%] Built target libuavcan_dsdlc
Scanning dependencies of target uavcan
[  3%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/driver/uc_can.cpp.o
[  7%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/marshal/uc_bit_array_copy.cpp.o
[ 11%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/marshal/uc_bit_stream.cpp.o
[ 14%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/marshal/uc_float_spec.cpp.o
[ 18%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/marshal/uc_scalar_codec.cpp.o
[ 22%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/node/uc_generic_publisher.cpp.o
[ 25%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/node/uc_generic_subscriber.cpp.o
[ 29%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/node/uc_global_data_type_registry.cpp.o
[ 33%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/node/uc_scheduler.cpp.o
[ 37%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/node/uc_service_client.cpp.o
[ 40%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/node/uc_timer.cpp.o
[ 44%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/protocol/uc_dynamic_node_id_client.cpp.o
[ 48%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/protocol/uc_node_status_provider.cpp.o
[ 51%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_can_acceptance_filter_configurator.cpp.o
[ 55%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_can_io.cpp.o
[ 59%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_crc.cpp.o
[ 62%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_dispatcher.cpp.o
[ 66%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_frame.cpp.o
[ 70%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_outgoing_transfer_registry.cpp.o
[ 74%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_transfer.cpp.o
[ 77%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_transfer_buffer.cpp.o
[ 81%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_transfer_listener.cpp.o
[ 85%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_transfer_receiver.cpp.o
[ 88%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/transport/uc_transfer_sender.cpp.o
[ 92%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/uc_data_type.cpp.o
[ 96%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/uc_dynamic_memory.cpp.o
[100%] Building CXX object libuavcan/CMakeFiles/uavcan.dir/src/uc_error.cpp.o
Linking CXX static library libuavcan.a
[100%] Built target uavcan
```

pavel-kirienko, 2015-08-15:
Thanks Ben. The fix has been merged into master.

closed 2015-08-15 by pavel-kirienko