OpenCyphal-Garage/libcyphal #56: Can't build and install library

jooptoorn, 2015-08-01:
Hi guys,
I can't seem to succesfully build the UAVCAN library on my linuxmint vm. When following instructions 
> mkdir build
>cd build
> cmake .. # Default build type is RelWithDebInfo, which can be overriden if needed.
> make
> sudo make install

the cmake step failed. First I manually installed gtest like described here, which fixed the issue:
http://stackoverflow.com/questions/13513905/how-to-properly-setup-googletest-on-linux

Then, the make step fails consistently and I have no clue what to do now.
> joop@yyy ~/libuavcan/build/libuavcan $ make
> Running from the source directory
> Traceback (most recent call last):
> File "dsdl_compiler/libuavcan_dsdlc", line 59, in <module>
>    from libuavcan_dsdl_compiler import run as dsdlc_run
>  File "/home/joop/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 17, in   module from uavcan import dsdl
> ImportError: No module named uavcan
> make[2]: *** [libuavcan/CMakeFiles/libuavcan_dsdlc] Error 1
> make[1]: *** [libuavcan/CMakeFiles/libuavcan_dsdlc.dir/all] Error 2
> make: *** [all] Error 2

I'm on kernel 3.13.0-37-generic
cmake version 2.8.12.2
python version 2.7

Because I was trying some stuff out on my new LPC11C24 board on Windows 7 x64 from LPCXpresso IDE, I cloned the uavcan into my workdir and followed these steps to get the example to compile: https://groups.google.com/forum/#!topic/uavcan/6-OG38bZDyE . The build fails with some errors:

>/bin/env: python: No such file or directory

> make: git: Command not found 

even though I have a working version of Git installed on Windows and Python 3.4 is added to path var, plus I can verify python version from Git bash.

> ../../../libuavcan/include/uavcan/time.hpp:12:32: fatal error: uavcan/Timestamp.hpp: No such file or directory

Reoccurs for many other files. I understand the uavcan library is to be built and installed before Timestamp.hpp is generated. If there is any way I could program the board with uavcan on Windows that would be just as helpful.

Thanks in advance

pavel-kirienko, 2015-08-01:
Hi!

First, you don't have to have gtest installed to build the library. It's only needed to make and run unit tests, i.e. it's only useful for library developers.

Second, regarding your Windows issues, you must install Python, either 2.7 or 3.x.

Third, your build fails on Mint because you forgot to initialize git submodules, like so: `git submodule update --init --recursive`. Please read the instructions at https://github.com/UAVCAN/libuavcan#cloning-the-repository.

pavel-kirienko, 2015-08-01:
Feel free to reopen if this didn't help.

closed 2015-08-01 by pavel-kirienko

jooptoorn, 2015-08-01:
Thanks for the quick reply! As it happens I forgot to cd into uavcan to initialize submodules the first trial, so the command failed, but I didn't notice. Sorry! Maybe put "cd uavcan" in there to make the instructions dummyproof?

I'll try to fix the python links somehow to make it work on Windows. Thanks for the help!

pavel-kirienko, 2015-08-01:
> Maybe put "cd uavcan" in there to make the instructions dummyproof?

Noted and fixed.

pavelvasev, 2018-04-01:
For some reason, system-wide installation in linux did produce same error: **No module named uavcan** when calling `libuavcan_dsdlc`. Only after performing the following commaind manually
```
libuavcan/libuavcan/dsdl_compiler/pyuavcan# ./setup.py install
```
the libuavcan_dsdlc compiler began to work.

pavel-kirienko, 2018-04-03:
That's probably because you're using a local venv. When you run `sudo pip`, the global instance of Python is affected; when you run just plain `python`, your user-specific instance of Python is used. If you have a solution or if you know how to improve the UX, please submit a pull request. Thanks!