OpenCyphal-Garage/libcyphal #64: Add Travis CI unit test configuration

bendyer, 2015-08-24:
May need to skip timing-dependent tests if build VM is not sufficiently performant.

pavel-kirienko, 2015-09-30:
It also makes sense to configure Travis to run Coverity Scan too.

See https://scan.coverity.com/travis_ci.

Instructions for manual build submission are here: https://github.com/UAVCAN/libuavcan#submitting-a-coverity-scan-build

bendyer, 2015-10-04:
Just looped back to this—please could you visit https://travis-ci.org/profile/UAVCAN and enable those three repos?

pavel-kirienko, 2015-10-04:
Done.

bendyer, 2015-10-04:
@pavel-kirienko re: Coverity Scan, I have sent you an access request which will hopefully allow me to download the sample Travis CI configuration referenced in step 8 of https://scan.coverity.com/travis_ci

pavel-kirienko, 2015-10-04:
Access granted.

bendyer, 2015-10-04:
No luck. Please could you try to access it and send me the file? Instructions are:

> 6. From the Coverity Scan Dashboard, click `Project Settings`. Then, on the right, click the `Submit build` button.
> 7. Assuming the project is properly registered via GitHub, you'll see a tab for `Configure Travis CI`. Visit that panel.
> 8. On the Travis CI Configuration page, you'll see a sample `.travis-yml` file. It will have auto-generated several of the necessary project-specific fields, including the encrypted Coverity Scan token (necessary to upload results).

pavel-kirienko, 2015-10-04:
OK, there's a problem with Coverity Scan indeed, I just emailed their tech support (you're CCed).

pavel-kirienko, 2015-10-07:
Coverity's benevolent tech support helped us to resolve the problem. Here's the file, @bendyer:

```yaml
env:
  global:
   # The next declaration is the encrypted COVERITY_SCAN_TOKEN, created
   #   via the "travis encrypt" command using the project repo's public key
   - secure: "nCQtEBd5gQ9zirHNQVpzB0Q8aVKmMuTrlHfZCK5ZOpFItD9zP0YwqUTcw6y8T+14CceY6Go/D+fgeMYR3QmYc2CW0vXMwgBYOFQuPYEef7455ZPdt6wdr4lnyP/B+fj4cWZ4WhV+hMigl2Ly4xaFH6msm+PlIHOdjdaXo3ko0LI="

addons:
  coverity_scan:
    project:
      name: "UAVCAN/libuavcan"
      description: "UAVCAN in C++"
    notification_email: pavel.kirienko@gmail.com
    build_command_prepend: "rm -rf build &> /dev/null || true; mkdir build && cd build && cmake .. -DCMAKE_BUILD_TYPE=Debug"
    build_command:   "make --ignore-errors"
    branch_pattern: coverity_scan
```

pavel-kirienko, 2015-10-10:
@bendyer I wonder if we could also build the LPC11C24 and STM32 demo applications with Travis. If the build server runs a recent enough Ubuntu, we could pull the toolchain from the repo.

antoinealb, 2015-10-10:
Travis runs Ubuntu 12.04, which means we can use the ARM embedded PPA (
https://launchpad.net/~terry.guo/+archive/ubuntu/gcc-arm-embedded). I
suggest separating the different builds using environment variables.

commit 84f68b91 (2015-10-24): Merge pull request #75 from UAVCAN/bendyer-issue-64

Travis CI configuration (resolves #64)
closed 2015-10-24 by pavel-kirienko