OpenCyphal-Garage/libcyphal #76: Travis CI can't execute unit tests properly because the environment is not real time enough

pavel-kirienko, 2015-10-24:
> https://github.com/UAVCAN/libuavcan/pull/75#issuecomment-150813549

@bendyer Nope, the build [has just failed](https://travis-ci.org/UAVCAN/libuavcan/jobs/87220991). There are actually two issues now:

1. Unit tests are failing because of real-time issues.
2. The build script redirects test outputs into log files, which are not accessible via Travis CI GUI.

A possible solution to 1 is to transition to the [new container-based Travis CI infrastructure](http://docs.travis-ci.com/user/migrating-from-legacy/?utm_source=legacy-notice&utm_medium=banner&utm_campaign=legacy-upgrade) - they are promising this:

> More available resources #
> The new containers have 2 dedicated cores and 4GB of memory, vs 1.5 cores and 3GB on our legacy infrastructure. CPU resources are now guaranteed, which means less impact from ‘noisy neighbors’ on the same host machine and more consistent build times throughout the day.

If that doesn't work, we'll have to resort to conditional disabling of real time tests.

As for 2, the build script has to detect when it's running on Travis CI, and conditionally disable unit test output redirection (I hope Travis can handle build outputs with ~30k lines). A possible solution is to add a CMake variable like `CONTINUOUS_INTEGRATION_BUILD` that will be set in `.travis.yml`. The same variable could also be used to implement the second part of the first solution, if need arises.

bendyer, 2015-10-24:
Option 1 would be difficult to orchestrate because we need to install updated software (Travis an ancient version of gcc by default). It's possible to store a bunch of binaries somewhere and install them without sudo, but that's getting pretty fragile.

Also, my local test environment is an 8-core VMware VM with 8 GiB RAM, and the tests fail on that (which is why I was surprised when it passed the first three times), so the container-based infrastructure may still not be enough.

I'll add the build flag and disable the timing-dependent tests when it's present.

bendyer, 2015-10-25:
Support code in https://github.com/UAVCAN/libuavcan/tree/issue-76

The test output can be downloaded from Travis.

None of the tests failed in that run (https://travis-ci.org/UAVCAN/libuavcan/builds/87256695) but I'll re-run it periodically and flag the failed tests as timing-dependent.

pavel-kirienko, 2015-10-26:
One just failed, it's `DynamicNodeIDClient.Basic`, see https://s3.amazonaws.com/archive.travis-ci.org/jobs/87427640/log.txt.

I'm sure it's not the only one.

pavel-kirienko, 2015-10-26:
Another one: `dynamic_node_id_server_NodeDiscoverer.RestartAndMaxAttempts`.

closed 2018-06-27 by pavel-kirienko