OpenCyphal-Garage/libcyphal #97: Baremetal STM32 driver is broken

pavel-kirienko, 2016-11-17:
We'll probably drop the baremetal support completely unless somebody submits an easy fix soon. The problem is that the file `chip.h` is missing, possibly something else.

davids5, 2016-11-17:
Hi Pavel,

My recollection is that it is for sitting on STLib to support Flow

I think there are just a few constants that were needed. 

https://github.com/PX4/Flow/blob/master/src/include/chip.h#L43-L47

pavel-kirienko, 2016-11-17:
Thanks David, let's see if somebody comes up with a fix, as I currently
have no bandwidth to test it myself.

JamesStewy, 2017-04-01:
Hi.

I have managed to get the baremetal STM32 driver working on an F303 discovery board and an F446 nucleo board. Here are the two `chip.h` files:

F3:

    #include "stm32f3xx.h"

    #define STM32F3XX
    #define STM32_PCLK1	(24000000ul)		// 24 Mhz
    #define STM32_TIMCLK1	(2*STM32_PCLK1)

F4:

    #include "stm32f4xx.h"

    #define STM32F4XX
    #define STM32_PCLK1		(42000000ul)		// 42 MHz
    #define STM32_TIMCLK1		(2*STM32_PCLK1)
    #define CAN1_TX_IRQHandler	CAN1_TX_IRQHandler	// This isn't defined anywhere
    #define CAN1_RX0_IRQHandler	CAN1_RX0_IRQHandler	// This isn't defined anywhere
    #define CAN1_RX1_IRQHandler	CAN1_RX1_IRQHandler	// This isn't defined anywhere

As mentioned above, I needed to include the header file for the specific device being used and set a few constants. I am using currently using the CubeMX graphical tool for development so I just read the value of `STM32_PCLK1` off the interface and `STM32_TIMCLK1` is always x2 of `STM32_PCLK1`. There is a function in the HAL library called `HAL_RCC_GetPCLK1Freq` which returns the same value.

The biggest issue I faced getting the bare metal driver working was linking up the interrupt handlers. For the F4 board, none of the CAN interrupt handlers where defined as preprocessor constants so I had to define them in `chip.h`. 

For the F3 board, the interrupt handlers are `USB_HP_CAN_TX_IRQHandler` (aliased as `CAN_TX_IRQHandler`), `USB_LP_CAN_RX0_IRQHandler` (aliased as `CAN_RX0_IRQHandler`) and `CAN_RX1_IRQHandler`. To get the F3 board working I had to make these changes to `uc_stm32_can.cpp`:

    @@ -22,7 +22,7 @@
     #endif
 
     #if (UAVCAN_STM32_CHIBIOS && CH_KERNEL_MAJOR == 2) || UAVCAN_STM32_BAREMETAL
    -# if !(defined(STM32F10X_CL) || defined(STM32F2XX) || defined(STM32F4XX))
    +# if !(defined(STM32F10X_CL) || defined(STM32F2XX) || defined(STM32F3XX)  || defined(STM32F4XX))
     // IRQ numbers
     #  define CAN1_RX0_IRQn USB_LP_CAN1_RX0_IRQn
     #  define CAN1_TX_IRQn USB_HP_CAN1_TX_IRQn
    @@ -64,6 +64,9 @@
     #define CAN1_TX_IRQn           CAN_TX_IRQn
     #define CAN1_RX0_IRQn          CAN_RX0_IRQn
     #define CAN1_RX1_IRQn          CAN_RX1_IRQn
    +#define CAN1_TX_IRQHandler     CAN_TX_IRQHandler
    +#define CAN1_RX0_IRQHandler    CAN_RX0_IRQHandler
    +#define CAN1_RX1_IRQHandler    CAN_RX1_IRQHandler
     #endif
     

I would like to help getting the bare metal driver working more universally but will need some guidance as to how.

James.

pavel-kirienko, 2017-04-01:
Great job James!

> I would like to help getting the bare metal driver working more universally but will need some guidance as to how.

I propose to add a new top-level section in the [STM32 platform driver manual](http://uavcan.org/Implementations/Libuavcan/Platforms/STM32/). Could you please [send your edits as a pull request](https://github.com/UAVCAN/uavcan.github.io/blob/master/Implementations/Libuavcan/Platforms/STM32/index.md)? Thanks in advance!

It would also help to add some quick reference (perhaps in the form of a link to the above doc page) in the comments near the place where `chip.h` is included.

> `STM32_TIMCLK1` is always x2 of `STM32_PCLK1`

Beware that this does not hold if the bit `TIMPRE` in `RCC_DCKCFGR` is set.

JamesStewy, 2017-04-01:
Is the plan that libuavcan won't include `chip.h` so every project will include their own and fill in the appropriate parameters?

pavel-kirienko, 2017-04-01:
Yes that's the plan.

JamesStewy, 2017-04-02:
Currently you might have to modify `uc_stm32_can.cpp` for each new chip so should the interrupt handler definitions be included in `chip.h`?

pavel-kirienko, 2017-04-02:
> should the interrupt handler definitions be included in chip.h?

I propose to support both approaches: provide definitions for popular and tested devices in `uc_stm32_can.cpp`, and let the user know that if their MCU is not supported, they will have to add definitions themselves in `chip.h`. I also suggest to include your extensions to `uc_stm32_can.cpp` described above into upstream.

closed 2017-04-02 by pavel-kirienko