OpenCyphal-Garage/libcyphal #108: dsdlc: C++ headers generated for union type definitions should be C++ unions themselves, not structs

pavel-kirienko, 2017-06-17:
Currently, the DSDL compiler turns DSDL union definitions into C++ structs, which causes the memory footprint to be a sum of that of the contained fields. This is really inefficient.

The DSDL compiler should use an aliased memory storage in order to reduce the size of the resulting type.

pavel-kirienko, 2019-03-17:
We should consider using `std::variant` instead for type safety reasons, as recommended by the C++ Core Guidelines. It is only available since C++17 though.

thirtytwobits, 2019-03-18:
Yeah, I don't think requiring C++17 or even 14 is a good idea just yet. C++11 is very widely supported but I'm aware of some RTOSes that have yet to implement C++14 or greater support.

pavel-kirienko, 2019-04-16:
On an unrelated note, this change is crucial for supporting the Register interface. Without aliasing, the footprint of `uavcan.register.Value` approaches 4 KiB:

https://github.com/UAVCAN/dsdl/blob/3baa2e90a0e8c7b660a2deda8f9a541877ed58e5/uavcan/register/Value.0.1.uavcan#L10-L27