OpenCyphal-Garage/libcyphal #116: Failing to build libuavcan

battlesnake, 2017-11-24:
When trying to build by following the build instructions in the README, or when running:

```
set -e
cd /tmp
git clone https://github.com/UAVCAN/libuavcan -b v1.0.0rc5 --recurse-submodules
cd libuavcan
# Unset flags in case my defaults are interfering
unset CFLAGS CXXFLAGS CPPFLAGS LDFLAGS MAKEFLAGS
# Set explicitly, to avoid using hardening wrapper
export PATH=/usr/bin
mkdir -p build
cd build
cmake ..
# Disable parallel build in case that's causing problems
make -j1
```

I see a load of warnings regarding `-Wexpansion-to-defined` for ifdef's in various headers, followed by fatal errors.

# Tail of build output including various warnings and the fatal error:

```
 #if UAVCAN_EXCEPTIONS
     ^~~~~~~~~~~~~~~~~
/tmp/libuavcan/./libuavcan/include/uavcan/protocol/logger.hpp:274:5: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/libuavcan/./libuavcan/include/uavcan/protocol/logger.hpp:274:5: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/libuavcan/./libuavcan/include/uavcan/protocol/logger.hpp:274:5: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/libuavcan/./libuavcan/include/uavcan/protocol/logger.hpp:274:5: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/libuavcan/./libuavcan/include/uavcan/protocol/logger.hpp:274:5: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/libuavcan/./libuavcan/include/uavcan/protocol/logger.hpp:274:5: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
/tmp/libuavcan/./libuavcan/include/uavcan/protocol/logger.hpp:274:5: warning: this use of "defined" may not be portable [-Wexpansion-to-defined]
In file included from /tmp/libuavcan/libuavcan_drivers/linux/include/uavcan_linux/uavcan_linux.hpp:10:0,
                 from /tmp/libuavcan/libuavcan_drivers/linux/apps/test_posix.cpp:7:
/tmp/libuavcan/libuavcan_drivers/linux/include/uavcan_linux/socketcan.hpp: In static member function ‘static can_frame uavcan_linux::SocketCanIface::makeSocketCanFrame(const uavcan::CanFrame&)’:
/tmp/libuavcan/libuavcan_drivers/linux/include/uavcan_linux/socketcan.hpp:64:105: warning: missing initializer for member ‘can_frame::__res0’ [-Wmissing-field-initializers]
         ::can_frame sockcan_frame { uavcan_frame.id& uavcan::CanFrame::MaskExtID, uavcan_frame.dlc, { } };
                                                                                                         ^
/tmp/libuavcan/libuavcan_drivers/linux/include/uavcan_linux/socketcan.hpp:64:105: warning: missing initializer for member ‘can_frame::__res1’ [-Wmissing-field-initializers]
/tmp/libuavcan/libuavcan_drivers/linux/include/uavcan_linux/socketcan.hpp:64:105: warning: missing initializer for member ‘can_frame::data’ [-Wmissing-field-initializers]
In file included from /usr/include/features.h:423:0,
                 from /usr/include/c++/7.2.0/x86_64-pc-linux-gnu/bits/os_defines.h:39,
                 from /usr/include/c++/7.2.0/x86_64-pc-linux-gnu/bits/c++config.h:533,
                 from /usr/include/c++/7.2.0/string:38,
                 from /tmp/libuavcan/./libuavcan/include/uavcan/build_config.hpp:138,
                 from /tmp/libuavcan/./libuavcan/include/uavcan/protocol/dynamic_node_id_server/event.hpp:8,
                 from /tmp/libuavcan/libuavcan_drivers/posix/include/uavcan_posix/dynamic_node_id_server/file_event_tracer.hpp:12,
                 from /tmp/libuavcan/libuavcan_drivers/linux/apps/test_posix.cpp:5:
/usr/include/bits/socket.h: In member function ‘int uavcan_linux::SocketCanIface::read(uavcan::CanFrame&, uavcan::UtcTime&, bool&) const’:
/usr/include/bits/socket.h:279:45: error: flexible array member ‘cmsghdr::__cmsg_data’ not at end of ‘struct uavcan_linux::SocketCanIface::read(uavcan::CanFrame&, uavcan::UtcTime&, bool&) const::Control’
     __extension__ unsigned char __cmsg_data __flexarr; /* Ancillary data.  */
                                             ^
In file included from /tmp/libuavcan/libuavcan_drivers/linux/include/uavcan_linux/uavcan_linux.hpp:10:0,
                 from /tmp/libuavcan/libuavcan_drivers/linux/apps/test_posix.cpp:7:
/tmp/libuavcan/libuavcan_drivers/linux/include/uavcan_linux/socketcan.hpp:225:48: note: next member ‘uint8_t uavcan_linux::SocketCanIface::read(uavcan::CanFrame&, uavcan::UtcTime&, bool&) const::Control::data [16]’ declared here
             std::uint8_t data[sizeof(::timeval)];
                                                ^
/tmp/libuavcan/libuavcan_drivers/linux/include/uavcan_linux/socketcan.hpp:222:16: note: in the definition of ‘struct uavcan_linux::SocketCanIface::read(uavcan::CanFrame&, uavcan::UtcTime&, bool&) const::Control’
         struct Control
                ^~~~~~~
/tmp/libuavcan/libuavcan_drivers/linux/include/uavcan_linux/socketcan.hpp:227:32: error: value-initialization of incomplete type ‘unsigned char []’
         auto control = Control();
                                ^
make[2]: *** [libuavcan_drivers/linux/CMakeFiles/test_posix.dir/build.make:63: libuavcan_drivers/linux/CMakeFiles/test_posix.dir/apps/test_posix.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:214: libuavcan_drivers/linux/CMakeFiles/test_posix.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

# Cmake output:

```
-- The C compiler identification is GNU 7.2.0
-- The CXX compiler identification is GNU 7.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using C++11 (pass UAVCAN_USE_CPP03=1 to override)
-- Release build type: RelWithDebInfo
-- Adding UAVCAN Linux platform driver
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Using uavcan target; source dir: /tmp/libuavcan/libuavcan
-- POSIX source dir: /tmp/libuavcan/libuavcan_drivers/posix
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/libuavcan/build
```

# Tool versions:

```
$ gcc --version
gcc (GCC) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ cmake --version
cmake version 3.9.6

CMake suite maintained and supported by Kitware (kitware.com/cmake).

$ lsb_release -a
LSB Version:    1.4
Distributor ID: Arch
Description:    Arch Linux
Release:        rolling
Codename:       n/a

$ uname -a
Linux hai 4.14.0-mark #1 SMP PREEMPT Mon Nov 13 16:50:57 GMT 2017 x86_64 GNU/Linux
```

pavel-kirienko, 2017-11-24:
Thanks. This code used to work with GCC 6, but apparently GCC 7 (according to your logs) doesn't like this:

https://github.com/UAVCAN/libuavcan/blob/fa6dbde51d2d4dac2a2b4eaed246453da2673dc9/libuavcan_drivers/linux/include/uavcan_linux/socketcan.hpp#L222-L226

On my system (GCC 6.3), `cmsghdr` is defined in `/usr/include/bits/socket.h` as follows:

```c
struct cmsghdr
  {
    size_t cmsg_len;            /* Length of data in cmsg_data plus length
                                   of cmsghdr structure.
                                   !! The type should be socklen_t but the
                                   definition of the kernel is incompatible
                                   with this.  */
    int cmsg_level;             /* Originating protocol.  */
    int cmsg_type;              /* Protocol specific type.  */
#if (!defined __STRICT_ANSI__ && __GNUC__ >= 2) || __STDC_VERSION__ >= 199901L
    __extension__ unsigned char __cmsg_data __flexarr; /* Ancillary data.  */
#endif
  };
```

`__cmsg_data` is a flex array, which is a non-standard extension in C++. It is supposed to contain the payload of the message we read from the kernel (which is a timestamp of type `::timeval`). I wrote an additional wrapper `Control` on top of `cmsghdr` in order to ensure type safety (flex arrays are obviously highly unsafe), which seems to be incompatible with newer versions of GCC.

If my understanding is correct, we could fix the error by removing the `Control` struct that I defined and using directly an instance of `cmsghdr`. However, the instance should be allocated in a way that would make the flex array extend to a valid memory region. We could accomplish that either with a union or we could just use [`std::aligned_storage`](http://en.cppreference.com/w/cpp/types/aligned_storage) or some similar alternative. (We could also use type punning to an array of bytes, but that violates the strict aliasing rules and therefore is unsafe)

@battlesnake would you be up to submitting a fix, please?

battlesnake, 2017-11-24:
The definition of `__glibc_c99_flexarr_available` is in [cdefs.h:136](https://code.woboq.org/userspace/glibc/misc/sys/cdefs.h.html#136), which is at `/usr/include/sys/cdefs.h` on my system.

So in order to prevent it from being defined, we would first need to include that header, then undefine, then include the other system headers.  Here, we're relying on the include guards to prevent it from being redefined, and as we're coupling our code to internal details of the system headers, it's starting to feel like quite an ugly hack.

One option might be something like:

```
union Control {
    struct cmsghdr hdr;
    struct {
        std::uint8_t _detail_hdr_skip[sizeof(struct cmsghdr)];
        std::uint8_t data[sizeof(::timeval)];
    };
};
```

battlesnake, 2017-11-24:
If that works, would it be acceptable?  If so, I'll create a PR

pavel-kirienko, 2017-11-24:
That looks okay. The following looks even simpler, in my opinion (you don't use the `struct` keyword in C++ when referring to a struct type):

```c++
union Control
{
    ::cmsghdr control;
    std::uint8_t _filler[sizeof(::cmsghdr) + sizeof(::timeval)];
};
```

Although the use of unions in high-integrity C++ is discouraged because they are inherently unsafe. The following is a preferred alternative:

```c++
static constexpr std::size_t ControlSize = sizeof(::cmsghdr) + sizeof(::timeval);
std::aligned_storage<ControlSize> control_storage;
::cmsghdr* control = new (&control_storage) ::cmsghdr();
```

battlesnake, 2017-11-24:
I had the dummy skip array in the nested struct in the union rather than just one array, so that the type would still present the `data` field as before.  The `std::aligned_storage` seems cleaner though.

battlesnake, 2017-11-24:
With the std::aligned_storage approach, it still fails to compile since the cmsghdr constructor attempts to initialise the flexible array.

battlesnake, 2017-11-24:
Is the initialisation of the structure necessary, given that we're receiving into it with `recvmsg`?

The simplest solution might be a uint8_t array.

pavel-kirienko, 2017-11-24:
Okay, then I propose to just zero-fill the aligned storage rather than invoking the constructor. What we're doing here is actually highly unsafe, but hey, it's POSIX.

battlesnake, 2017-11-24:
Ok.  I've implemented it.  Also to note - `std::aligned_storage<ControlSize>` (from your example) is intended as a trait rather than a type.  I'll send a PR now, which uses `std::aligned_storage` and `std::fill`, a `reinterpret_cast` to cast the storage to a byte array, then `std::fill` to initialise it.

pavel-kirienko, 2017-11-24:
Cool. You're right, in my example above I should have used either `aligned_storage_t` or `aligned_storage::type`

PR #117 (2017-11-24): Fixed failing build when socket.h adds flexible arrays to end of structures
Fixes #116
closed 2017-11-24 by pavel-kirienko