OpenCyphal-Garage/libcyphal #121: Linker issues with arm-none-eabi-g++ 5.4.1

nfriendly, 2018-02-02:
I'm integrating libuavcan into a project and my linker keeps segfaulting after I've added libuavcan into my build. I've determine the offending portion of code is `initialize(int)` method of `ArrayImpl` at line 296 of [include/uavcan/marshal/array.hpp](https://github.com/UAVCAN/libuavcan/blob/master/libuavcan/include/uavcan/marshal/array.hpp#L296). I haven't figured out exactly what occurs, but removing this method solves the problems. Newer versions of the compiler+linker successfully compiles the code into a functioning binary.

Could you explain the intent of the of template `typename EnableIf<sizeof(U(0) >= U())>::Type initialize(int)`? It appears you are trying to define a method that constructs all members of an array if a default constructor exists, and does nothing if the default constructor does not exist. What I don't understand is the `U(0) >= U()` portion. I believe simply writing `sizeof(U())` inside the `EnableIf` template would accomplish this goal. What does the `U(0)` provide?

pavel-kirienko, 2018-02-03:
I am tempted to suggest to throw your linker out the window. The below text is written on the assumption that that is not an option.

The intent of that template is to check if the `ValueType` is a scalar type using SFINAE. If it is, the array will be zero-filled. If it is not, it is assumed that the `ValueType` defines a default constructor that will bring each item into proper state upon construction. `U()` checks the presence of a default constructor, whereas `U(0)` checks the presence of a scalar constructor that accepts zero. The comparison operator `>=` further validates whether `ValueType` is comparable. This simple check allows the array class to discern complex types used in the library from scalar types.

Perhaps you could somehow rearrange the check to keep the linked happy? Maybe the check could be moved into an external struct, e.g. `IsScalarType<U>::Value` or whatever.

pavel-kirienko, 2018-05-07:
Closing as stale, please re-open if this is still relevant.

closed 2018-05-07 by pavel-kirienko