OpenCyphal-Garage/libcyphal #124: Uavcan initialization failed??

programdog, 2018-04-26:
$ make posix_rpi_native.cmake

error log:
```
wzh> uavcan start
WARN  [uavcan] Node ID 1, bitrate 1000000
WARN  [uavcan] power sub success ! res=0
WARN  [uavcan] monitor sub success ! res=0
WARN  [uavcan] rgbled sub success ! res=0
ErrInvalidConfiguration=13
WARN  [uavcan] CentralizedServer init: -13
WARN  [uavcan] Node init failed: -13
Command 'uavcan' failed, returned -13
```

Ask for help !!!

```
 
1. /*
2.          * Making sure that the server is started with the same node ID
3.          */
4.         {
5.             const NodeID stored_own_node_id = storage_.getNodeIDForUniqueID(getOwnUniqueID());
6.             if (stored_own_node_id.isValid())
7.             {
8.                 if (stored_own_node_id != node_.getNodeID())
9.                 {
10.                     printf("ErrInvalidConfiguration=%d\r\n",ErrInvalidConfiguration);
11.                     return -ErrInvalidConfiguration;
12.                 }
13.             }
14.             else
15.             {
16.                 res = storage_.add(node_.getNodeID(), getOwnUniqueID());
17.                 if (res < 0)
18.                 {
19.                      printf("storage_.add=%d\r\n",res);
20.                     return res;
21.                 }
22.             }
23.         }

```

pavel-kirienko, 2018-04-26:
Looks like the node ID of your server has changed. That means that you have to wipe the allocator's data directory.

pavel-kirienko, 2018-04-26:
Please reopen if the above did not help.

closed 2018-04-26 by pavel-kirienko