OpenCyphal-Garage/libcyphal #126: why Multiple repeated calls to the sending function ，then%cpu : 90%+

programdog, 2018-05-02:
in px4 , like rgbled  . i  make  a moudle : uavcan_rgbled ,
I use : work_queue(LPWORK, &_work, (worker_t)&UavcanRgbled::led_trampoline, this, _led_interval/8);

-----------------------------------------------------------------------------------------------
this Call flow : (When it's time to call again and again)
led_trampoline() 
                         ->led()->send_led_rgb()
                                                             ->_rgbled_pub.broadcast(msg)
------------------------------------------------------------------------------------------------

but Random time （1~5min）：look at the CPU  90%+!!
 PID     USER     PR  NI    VIRT    RES    SHR S    %CPU    %MEM     TIME+
123117 root      20   0  170276  39112   4660 S    94.4       1.9   0:10.33

Why did this happen?  thank you for your help

pavel-kirienko, 2018-05-02:
Let's have a look at the code.

programdog, 2018-05-03:
I compressed 4 file ：uavcan_main.cpp , uavcan_main.hpp , rgbled.cpp , rgbled. hpp 

in **uavcan_main.cpp** in line 665 only use command of "rgbled" to  get into “rgbled_test()” then use RGBLED_MODE_BLINK_FAST ，then get into rgbled.cpp to handle。

in rgbled.cpp ，Function jump like ：

 uavcan_rgbled_set_mode(RGBLED_MODE_BLINK_FAST) -》set_mode(); -》 work_queue(); -》led_trampoline() -》led()-》work_queue() 

then When it's time to call again and again when i use function Flicker 

but Random time （1~5min）：look at the CPU 90%+!!
PID USER PR NI VIRT RES SHR S **%CPU** %MEM TIME+
123117 root 20 0 170276 39112 4660 S **94.4** 1.9 0:10.33

Why did this happen? thank you for your help

[uavcan_rgbled.zip](https://github.com/UAVCAN/libuavcan/files/1969405/uavcan_rgbled.zip)

programdog, 2018-05-05:
-_-!

programdog, 2018-05-07:
[~_~ ]

programdog, 2018-05-07:
~_~#

programdog, 2018-05-07:
=_=*

pavel-kirienko, 2018-05-07:
You don't initialize the field `_pattern.duration`. It remains at zero after the memset call in the constructor, leading to continuous requeuing of your handler.

Please note that this is not an appropriate place for general programming questions.

closed 2018-05-07 by pavel-kirienko

programdog, 2018-05-08:
thankyou for your help , I'll try again。It may not be a code problem ，I will keep finding，then I will tell you what i am finding.