OpenCyphal-Garage/libcyphal #131: libuavcan build issues.

Tal-eigen, 2018-06-05:
Hi,
This is the compile error g++ throws
```
outgoing_transfer_registry.hpp:89:34: error: 'const class uavcan::OutgoingTransferRegistryKey' has no member named 'toString'
                              key.toString().c_str(), int(value.tid.get()));

```
```
src/libuavcan/libuavcan/src/protocol/uc_node_status_provider.cpp:109:86: error: no matching function for call to 'uavcan::MonotonicDuration::toString()'
                  period.toString().c_str(), node_status_pub_.getTxTimeout().toString().c_str());
```

Since i am compiling this for stm32. I havent defined the UAVCAN_TOSTRING macro.
 Is there a workaround for this?. Below is the Cmake File which contains libuavcan subdirectory. 

```
cmake_minimum_required(VERSION 3.2 FATAL_ERROR)
                               
set(CMAKE_MODULE_PATH ${TOOLCHAIN_PATH})
set(PROJECT_CONF_DIR ${CMAKE_CURRENT_LIST_DIR}/stm32_uavcan/conf)

set(STM32_FAMILY F4 CACHE INTERNAL "stm32 family")

find_package(Chibios REQUIRED rt hal)

  
option(LIBUAVCAN_SUPPORT "Build with libuavcan" OFF)

#Libuavcan specific defines
  add_definitions(
                  -DUAVCAN_STM32_TIMER_NUMBER=3
                  -DUAVCAN_STM32_NUM_IFACES=1
                  -DUAVCAN_STM32_CHIBIOS=1
                  -DUAVCAN_CPP_VERSION=UAVCAN_CPP11
                  -DUAVCAN_STM32_IRQ_PRIORITY_MASK=4
                  )

if(LIBUAVCAN_SUPPORT)
  set(UAVCAN_PLATFORM stm32 CACHE STRING "uavcan stm32 platform")
  add_subdirectory(libuavcan)
endif()

add_subdirectory(stm32_uavcan)
```

pavel-kirienko, 2018-06-05:
Apparently you have `UAVCAN_DEBUG` enabled. This option enables `UAVCAN_TRACE()` here: https://github.com/UAVCAN/libuavcan/blob/89e68cfa47dc0a5420358204b046f3aee043f9fa/libuavcan/include/uavcan/debug.hpp#L19 which in turn pulls in your `toString()` here: https://github.com/UAVCAN/libuavcan/blob/1e60063e8209195837b740d1e32c0f6b0a4ca4a5/libuavcan/src/protocol/uc_node_status_provider.cpp#L108-L109

The solution is to disable `UAVCAN_DEBUG`.

Tal-eigen, 2018-06-05:
yup. thanks

closed 2018-06-05 by Tal-eigen