OpenCyphal-Garage/libcyphal #136: Linux CLI tools don't support --version or --help

edwinhayes, 2018-06-13:
The Linux CLI tools don't support common arguments for getting help or version information, except for `uavcan_add_vcan`, which handles `--help` but not `--version`.

1. It would be helpful to be able to determine the version number of the library which is installed on a given machine.

2. I can't be the only person who wouldn't have expected that `uavcan_add_vcan --version` was actually going to create a new interface of that name!

pavel-kirienko, 2018-06-13:
I would happily accept PRs fixing these shortcomings. Right now these are low-priority issues.

edwinhayes, 2018-06-13:
Sure.  Am short staffed at the moment, but hopefully I'll be able to get someone to do this at some stage.

closed 2024-04-04 by pavel-kirienko