OpenCyphal-Garage/libcyphal #144: Add a .clang-format that enforces Zubax coding conventions

thirtytwobits, 2018-07-04:
.clang-format should enforce as much of https://kb.zubax.com/pages/viewpage.action?pageId=2195699 as is supported by that technology.

Protip: Use a tool like https://clangformat.com/ to help.

antoinealb, 2018-07-04:
There is already an uncrustify config for the Zubax style guide apparently: https://kb.zubax.com/pages/viewpage.action?pageId=2195703

pavel-kirienko, 2018-07-04:
Uncrustify sucks. I heard people say that clang-format sucks considerably less, but I had no chance to verify that myself yet.

SyrianSpock, 2018-07-04:
Using clang-format (6.0) for the last few months at work (C++ code base), and it works way better than uncrustify in my experience. The config file is also much easier to setup.

thirtytwobits, 2018-07-05:
Yeah, clang-format is emerging as a defacto standard for code formatting. It's supported everywhere now

closed 2018-07-22 by pavel-kirienko