OpenCyphal-Garage/libcyphal #147: Remove all dead code within UAVCAN_CPP_VERSION < UAVCAN_CPP11 macros

thirtytwobits, 2018-07-15:
search for and remove all macros that generate code UAVCAN_CPP_VERSION < UAVCAN_CPP11 since this statement will never be true after c++03 support was dropped.

pavel-kirienko, 2018-07-16:
We should be careful not to break compatibility with NuttX in the process. In the C++11 mode, libuavcan assumes that [the entirety of the standard library is available](https://uavcan.org/Implementations/Libuavcan/#c-compiler-requirements), whereas the C++03 mode is equipped with certain hacks made to work around the bugs and missing features in the NuttX standard library. Although the state of the standard library there could have been improved since then, not sure. 

The point is, we should probably set up some NuttX-based testing environment. Do you have a NuttX-compatible CAN FD-enabled demo board like NUCLEO-H743ZI?

kjetilkjeka, 2018-07-16:
I assume these errors will happen at linking? Is travis building for NuttX? If so, can this be detected in CI?

pavel-kirienko, 2018-07-16:
> I assume these errors will happen at linking?

Better, they happen at compiling.

> Is travis building for NuttX?

Nope :( We need a test target for that. Like some kind of a test application written for some sort of Nucleo.

thirtytwobits, 2019-05-04:
v0 will not be updated, wholesale, to c++11. This will happen as part of the v1 migration.

closed 2019-05-04 by thirtytwobits