OpenCyphal-Garage/libcyphal #153: NuttX testing target

pavel-kirienko, 2018-07-22:
@LorenzMeier @davids5 @dagar Is it possible to get some help with this?

dagar, 2018-07-23:
What do you mean exactly? If it's actual automated testing with real hardware I'd be interested in helping, but ideally it would overlap with something that can also run PX4.

davids5, 2018-07-23:
@pavel-kirienko - While H7 is now on master in NuttX. Unfortunately Greg removed libuavcan support after it failed to build. PX4 is still on an older branch and the upgrade will require all the board configs to be revetted.   So there is no quick path to get a test fixture up and running here.

pavel-kirienko, 2018-07-23:
We don't really need a built-in libuavcan support in NuttX. What we need is a simple application that instantiates a UAVCAN node using libuavcan, broadcasts some simple messages and invokes/serves some services. The objective is to ensure that the library compiles against relatively modern versions of NuttX. Unless they managed to fix their C++ standard library, Libuavcan is going to require certain NuttX-specific hacks to make it work there (it is also the reason why it broke after the last upgrade I suppose). Being able to test that with real hardware in an automated fashion is a huge plus, but not strictly required at the moment. The test application can also be probably based on PX4 if that helps.

closed 2024-04-04 by pavel-kirienko