OpenCyphal-Garage/libcyphal #157: Compiler errors on missing files in uavcan/protocol directory

mjs513, 2018-08-03:
Hi.  Really new to this and managed to get it into a format for use in Teensy 3.5 sketches using the Arduino IDE.  But as I am going through it to see what has to change if anything I was getting a bunch of errors on missing files used in the include statement's.  All the errors were in the protocol directory:

1. GetDataTypeInfo..hpp not found - referenced in data_types_info_provider.hpp
2. There if no "file" directory and associated files referenced in file_server.hpp
3. GlobalTimeSync.hpp not found - referenced in global_time_sync_slave.hpp
4. No "debug" directory
5. GetNodeInfo.hpp not found ref. in node_info_retreiver.hpp
6. Node_Stats.hpp not found - ref in node_stats_monitor.hpp
7. Panic.hpp not found - panic_broadcaster.hpp
8 No "param" directory of associated files
9. no Restart_Node.hpp - ref in restart_request_server.hpp
10. No Get_Transport_stats.hpp - ref in transport_stats_provider.hpp.

Any help would be appreciated.

Thanks

Mike

pavel-kirienko, 2018-08-03:
The missing files are supposed to be auto-generated by the DSDL compiler. An example of its invocation can be found in the CMake files or here https://github.com/UAVCAN/libuavcan_stm32#building or just run [`libuavcan_dsdlc --help`](https://github.com/UAVCAN/libuavcan/blob/master/libuavcan/dsdl_compiler/libuavcan_dsdlc)

closed 2018-08-03 by pavel-kirienko

mjs513, 2018-08-03:
Yep.  They were generated.  Just didn't realize they were autogenerated. 
Thanks