OpenCyphal-Garage/libcyphal #159: IO latency is bad

pavel-kirienko, 2018-08-07:
Evidence from @bugobliterator: https://github.com/ArduPilot/ardupilot/pull/8987#issuecomment-409235227 (bad buffering architecture leads to slow field serialization and deserialization)

From @mhkabir:  

> Also after some instrumentation (not related to Myxas directly), I found :
>
> `uavcan control latency: 1268 events, 377221us elapsed, 297.49us avg, min 165us max 2440us 322.559us rms` (Myxa PX4 copter)
>
> `io control latency: 47776 events, 8705549us elapsed, 182.22us avg, min 160us max 291us 49.261us rms` (Normal PX4 copter)
>
> The worst case 2ms latency is b a d.

This is what I had to say on the subject (copy from the linked thread above):

> The linked list based buffer requires sequential iteration of the buffer segments in order to reach the desirable offset, which makes it slow. Additionally, the code that you identified as the culprit makes a lot of auxiliary operations per every byte read in order to facilitate the segment traversal logic at the outer layer. One could get rid of the list traversal problem by using some sort of a stateful reader object which would keep references to the current chunk. Is anybody willing to give it a try?

The culprit can be easily found in the sources by grepping for the key phrase `This shall be optimized`. As it stands, I wrote it back in 2014.

[image]

pavel-kirienko, 2018-08-09:
Relevant observations: https://github.com/PX4/Firmware/issues/10171#issuecomment-411669947

thirtytwobits, 2018-09-11:
I have a fix in the works for this. Stay tuned.

closed 2018-10-02 by pavel-kirienko