OpenCyphal-Garage/libcyphal #173: DSDL compilation

SergeyF1987, 2018-10-29:
I feel so stupid.. But I cannot understand how can I do DSDL compilation.
So... I get messages like "Timestamp.hpp" - no such file or directory. So, I read about this at 
https://uavcan.org/Implementations/Libuavcan/FAQ/
"This could be caused by failing DSDL compilation - see below."
But no any usefull information below
As I understood I must launch command in pre-build string in Eclipse like:
C:\Python36\python.exe  D:\CANtest\EmptyCAN2\libuavcan\libuavcan\dsdl_compiler\libuavcan_dsdlc
But... Nothing happens..  I cannot compile and find "Timestamp.hpp" or other needed hpp-files

thirtytwobits, 2018-10-29:
You have to supply the path to the DSDL files. In libuavcan's master branch this is a submodule mounted under the "dsdl" directory.

pavel-kirienko, 2018-11-01:
Here: https://github.com/UAVCAN/libuavcan_stm32#using-make look for the text "Invoke DSDL compiler and add its default output directory to the include search path"

closed 2018-11-01 by pavel-kirienko

pavel-kirienko, 2018-11-01:
Wrote a few sentences about libuavcan_dsdlc here: https://uavcan.org/Implementations/Libuavcan/#dsdl-compiler