OpenCyphal-Garage/libcyphal #174: node.start() is freezes

SergeyF1987, 2018-10-31:
Another trouble appeared. I build my project for STM32 without any mistakes. 
So I initialized my single node like in manual:
https://uavcan.org/Implementations/Libuavcan/Tutorials/2._Node_initialization_and_startup/
It's right and has not any troubles.
//--------------------------------------
	node_id = 1;
    auto& node = getNode();
	node.setNodeID(node_id);
	node.setName("AirSensor");
	node.start();
	uavcan::Publisher <uavcan::protocol::NodeStatus> pub(node);
	pub.init();
    pub.setPriority(uavcan::TransferPriority::MiddleLower);
	node.setModeOperational();
//------------------------------------------

But programm stops on node.start(). I spied this hell in debugger and found that program freezes at SentToIface function in uc_can_io.cpp (string 231):
//---------------------------------------------------
    ICanIface* const iface = driver_.getIface(iface_index);
    if (iface == UAVCAN_NULLPTR)
    {
        UAVCAN_ASSERT(0);   // Nonexistent interface  ----> This macros freeze the programm
        return -ErrLogic;
    }
//---------------------------------------------------
Actually  UAVCAN_ASSERT(0); freeze all platform. And no any error returns.
So, driver_.getIface(iface_index); always returns 0 at first, because iface_index variable we get from -->
 res = sendToIface(i, frame, tx_deadline, flags); where i get from cycle (for (i=0.. bla-bla-bla))
And.. driver_.getIface(iface_index) goes to:
//---------------------------------------------------
CanIface* CanDriver::getIface(uavcan::uint8_t iface_index)
{
    if (iface_index < UAVCAN_STM32_NUM_IFACES)
    {
        return ifaces[iface_index];
    }
    return UAVCAN_NULLPTR;
}
//-----------------------------------------------------------
I have UAVCAN_STM32_NUM_IFACES=1 and as I said I always get 0 value at first, and it returns ifaces[0]=0. Then it goes to UAVCAN_ASSERT(0); that completely freezes all programm (like in eternal while-cycle)

I cannot understand what happens?? How init nodes on STM32 properly??

antoinealb, 2018-10-31:
The STM32 implementation of the drivers can use several "backends" depending on your RTOS. Did you define one of `UAVCAN_STM32_CHIBIOS`, `UAVCAN_STM32_NUTTX`, `UAVCAN_STM32_BAREMETAL` or `UAVCAN_STM32_FREERTOS`?

You should also define `UAVCAN_STM32_TIMER_NUMBER` to a timer that can be freely used by UAVCAN.

SergeyF1987, 2018-10-31:
Of course, I defined it in build_config.hpp:
# define UAVCAN_STM32_BAREMETAL 1
.....
# define UAVCAN_STM32_TIMER_NUMBER 7
....
and #define UAVCAN_STM32_NUM_IFACES 1
..
But problem still exist. Actually I only commented  UAVCAN_ASSERT(0); ( this freezes, so i think that it's compilator's feauture) and program works further, but returns ErrLogic at string below.
Maybe I forgot something else, or this driver works not properly with STM32F407..?! (try to launch it on Discovery-board)

antoinealb, 2018-10-31:
Hmm thats weird. I am using UAVCAN in an F407 without any issue. I just noticed that TIM7 is a "basic timer", I am currently using TIM2. Could you try with that one to see if that works please?

SergeyF1987, 2018-10-31:
Yes. I use this driver. So I compiled DSDL-directory and included it in project. So I directly defined UAVCAN_STM32_TIMER_NUMBER , UAVCAN_STM32_NUM_IFACES and ets. in build_config.hpp at stm32 driver. I dont used 'make' command. So now I think, maybe it's causes the trouble?!

SergeyF1987, 2018-10-31:
I see, that this func returns ifaces[0]=0, that call ErrLogic issue:
CanIface* CanDriver::getIface(uavcan::uint8_t iface_index)
{
    if (iface_index < UAVCAN_STM32_NUM_IFACES)
    {
        return ifaces[iface_index]; //returns 0, that calls error
    }
    return UAVCAN_NULLPTR;
}
In debugger I also found that:
1) Failed to execute MI command:
-var-create - * &(((ifaces[0])->pending_tx_))
Error message from debugger back end:
-var-create: unable to create variable object
It seems that programm cannot set unzero-value to ifaces[x]. I still cannot understans, why.
I use Atollic TRUEstudio project, generated with CubeMX

SergeyF1987, 2018-11-01:
What IDE your use?

SergeyF1987, 2018-11-02:
This topic is closed, I fixed troubles. The reason was simple. I initialized CAN bus twice (in HAL first, and in libuavcan second), that caused conflict. And node init-process ended with error.  But I cannot understand why assert(x) macro go to frost. But now it's not so matter for me.

closed 2018-11-03 by pavel-kirienko