OpenCyphal-Garage/libcyphal #176: UTC is a misnomer

pavel-kirienko, 2018-11-01:
There are two types of clock in libuavcan:

* Monotonic clock, which only goes forward steadily at the same rate (with possible minor rate deviations caused by the instability of the base oscillator), and never leaps.
* Synchronized clock, which can leap and change rate in order to keep the local time base synchronized with external sources.

Now, the current naming convention is as follows: the monotonic clock is called "monotonic", as one would expect, but the synchronized clock is called "UTC". I guess my mind was in a serous disarray when I came up with that naming; we have to do something about that.

I suggest replacing "UTC" with "Synchronized", and keep "UTC" as a deprecated alias for a while.

antoinealb, 2018-11-01:
I don't have a strong opinion on this, but as a point of data, POSIX uses "Monotonic" and "Realtime" to describe those two types of clocks: https://linux.die.net/man/3/clock_gettime

kjetilkjeka, 2018-11-02:
Realtime may be misleading. The rust stdlib uses "SystemTime" for this type of clock, I think that conveys more information and is less confusing than "Syncrhonized"

antoinealb, 2018-11-02:
I think SystemTime is pretty good indeed.

pavel-kirienko, 2018-11-02:
> "SystemTime" [...] conveys more information and is less confusing than "Syncrhonized"

I would argue the direct opposite of that. System time is far too generic, could mean anything, including the monotonic time (remember this is not a local OS we're talking about, but a protocol). If we fail to agree on this, how about saying "Wall time"? This term is used in ROS.

SergeyF1987, 2018-11-02:
Synctonic will be lol)

pavel-kirienko, 2018-11-03:
After some extra consideration I ended up agreeing with Kjetil that System Time might be an acceptable option. It is used in the same sense in other sources, including standard libraries of C++ or Wolfram Mathematica. It should also be acceptable to extend the notion of "system" from just the local machine running a UAVCAN node to the entirety of the bus.

Unless there are objections, let's choose Monotonic and System.

antoinealb, 2018-11-03:
> It should also be acceptable to extend the notion of "system" from just the local machine running a UAVCAN node to the entirety of the bus.

Actually that makes me realize, why not call them `Monotonic` and `NetworkTime` (or `BusTime`). That being said, I am OK with System.

thirtytwobits, 2019-05-29:
Do we want to fix this in v0 as well @pavel-kirienko ?

pavel-kirienko, 2019-05-29:
I don't think we should do that.

closed 2024-04-04 by pavel-kirienko