OpenCyphal-Garage/libcyphal #178: Refactor libuavcan to be a header-only library.

thirtytwobits, 2018-11-08:
This reduces the opportunity for errors that can arise when the library objects are compiled with different flags than the headers used by an application. Given that libuavcan is already heavily templated the opportunity for such errors is high.

antoinealb, 2018-11-08:
Wouldn't that massively increase the compile time of the application though ?

pavel-kirienko, 2018-11-08:
Not much, because, as Scott said, libuavcan is heavily templated as it is; these templates will stay where they are. The bits of code that are currently defined in the `.cpp` units are insignificantly small. Even if that were not the case, who cares about compilation time? Its increase is not that significant compared to the advantages of a simpler usage and a reduced risk of errors; besides, it can be fixed by switching to a faster dev hardware.

Another advantage of a header-only library not mentioned above is that the compiler will have an opportunity to discard dead code from the final binary, improving compliance with certain high-reliability coding standards.

thirtytwobits, 2018-11-08:
I also just lost a day of productivity working with a build system that was compiling a dot-a of libuavcan using one set of flags and then using those objects with headers compiled with a different set of build flags. Insidious bugs ensued.

I'll soften Pavel's statement though: I care very much about build time but I don't think this change will significantly increase overall build time.

antoinealb, 2018-11-08:
>  I care very much about build time but I don't think this change will significantly increase overall build time.

Okay, I don't know enough about how C++ build time scales to say that with confidence, hence my question. If you guys say we should be fine, that's OK for me.

thirtytwobits, 2020-02-21:
Closing this since the task is basically "finish v1"

closed 2020-02-21 by thirtytwobits