OpenCyphal-Garage/libcyphal #191: The library documentation should be located in its source repository (here)

pavel-kirienko, 2018-12-19:
Currently, the Libuavcan docs *(btw what's the right way to spell it -- libuavcan or Libuavcan? One would think that I should know but I still have no idea)* are located on the website: https://uavcan.org/Implementations/Libuavcan

I think moving them to the website was a mistake as it decouples the versioning processes of the docs and of the library, which are actually rightly related and rather should be managed synchronously. I propose that we should create a directory here and move all docs from the website there.

thirtytwobits, 2018-12-20:
Strongly agree with this. 

(I've always used lowercase l since that's how linux names .so libraries. That said, English rules would uppercase it as a proper name. So... I don'no either.)

pavel-kirienko, 2019-11-20:
I think this is already fixed for v1 and for v0 we are going to leave it as-is. When I opened this issue, I meant it to be related to v1.

closed 2019-11-20 by pavel-kirienko