OpenCyphal-Garage/libcyphal #194: 'Reference is not a tree' with old commit

dominiknatter, 2019-01-19:
Hi everyone,

I wanted to clone a repository which used libuavcan pointing at [`e1c9a4f`](https://github.com/UAVCAN/libuavcan/commit/e1c9a4f5064a336295125da25c0542a5cd601dd7) as a submodule. However, I got an error. I tried to investigate further: I cloned the most recent version (which worked) with the goal to checkout the desired SHA. However, I got the same error, see here:

[image]

Does this somehow link to your already closed [issue](https://github.com/UAVCAN/libuavcan/issues/139)? The fix in there is not found anymore. Do you have any suggestions on how to solve this problem?

dominiknatter, 2019-01-19:
As a remark: I was able to clone the repository mentioned by @dagar in [#139](https://github.com/UAVCAN/libuavcan/issues/139), i.e. [this one](https://github.com/dagar/libuavcan/tree/master_upstream_nuttx_irq_changes), which looks like it could be identically the same.

pavel-kirienko, 2019-01-19:
A few days ago I asked @davids5 to delete branches that are no longer needed, which he did (thanks), and that is probably the reason you're seeing the error. Perhaps you should update your submodule to a newer commit?

dominiknatter, 2019-01-23:
Hi @pavel-kirienko,

the repository was updated and now uses the newest commit successfully.

closed 2019-01-23 by pavel-kirienko