OpenCyphal-Garage/libcyphal #195: Consider simplifying the TX queue management

pavel-kirienko, 2019-01-22:
Inspired by https://forum.uavcan.org/t/alternative-transport-protocols/324 (search for *"prevent stale data from backing up in the transmit queue, because if a connection is restored, the stale data would be released to the network and possibly disrupt..."*) and https://github.com/UAVCAN/libuavcan/issues/189#issuecomment-455884496 (thanks @Zarkopafilis).

The current TX queue maintenance logic is quite complex and resource-consuming. It is intended to prevent a likely service disruption when one of the redundant interfaces is temporarily disconnected and then reconnected back (as explained in the first linked article above). It could be simplified.

Instead of using a sophisticated QoS management logic with full-queue traversal on overflow, I propose that we simplify the logic as follows:

* When adding a new item, check if the queue is full.
  * If the queue is full, abort immediately and drop the new frame, instead of attempting to replace an existing older frame in the queue. This will not affect a properly designed bus unless the interface is disconnected, because a properly designed bus provides latency guarantees by definition. We don't care about disconnected interfaces because they are unusable anyway.
  * If the queue is not full, proceed as usual: add the frame at the correct position according to its CAN ID priority.
* When removing an item, search for the highest-priority one (it is done this way currently, no change). When found, check the expiration timeout.
  * If the frame is expired, drop it instead of transmitting it, and try the next one. This condition will prevent old frames from disrupting the bus when the interface is re-connected after a temporary failure.
  * If the frame is not expired, transmit it and remove it from the queue as usual.

This change will drastically simplify the logic, reduce the ROM footprint and processing load, render QoS unnecessary, and fix the determinism issues. We still want an *O(log n)* queue though, so #189 still applies.

thirtytwobits, 2019-01-23:
> I was in the process of writing tests

👍

pavel-kirienko, 2019-03-06:
Fixed in #198

closed 2019-03-06 by pavel-kirienko