OpenCyphal-Garage/libcyphal #203: Boolean saturated to 0 length

spiremike, 2019-02-02:
Using DSDL definition `dsdl/101.Bool.uavcan`:
```
bool in
---
bool out
```

The following is generated:
```
uint32_t dsdl_BoolRequest_encode_internal(dsdl_BoolRequest* source,
  void* msg_buf,
  uint32_t offset,
  uint8_t CANARD_MAYBE_UNUSED(root_item))
{
    source->in = CANARD_INTERNAL_SATURATE_UNSIGNED(source->in, 0)
    canardEncodeScalar(msg_buf, offset, 1, (void*)&source->in); // 0
    offset += 1;

    return offset;
}
```

Saturation to length zero forces the saturated value to always be 0.

No longer applying saturation, and treating bool as an unsigned type for max size calculation seems to fix the issue, apply the diff:
```
diff --git a/dsdl_compiler/libcanard_dsdl_compiler/__init__.py b/dsdl_compiler/libcanard_dsdl_compiler/__init__.py
index 9b66128..474a8d3 100644
--- a/dsdl_compiler/libcanard_dsdl_compiler/__init__.py
+++ b/dsdl_compiler/libcanard_dsdl_compiler/__init__.py
@@ -217,9 +217,9 @@ def type_to_c_type(t):
                     'post_cpp_type':'',
                     'cpp_type_comment':'bit len %d' % (t.bitlen, ),
                     'bitlen':t.bitlen,
-                    'max_size':get_max_size(t.bitlen, False),
+                    'max_size':get_max_size(t.bitlen, True),
                     'signedness':signedness,
-                    'saturate':saturate}
+                    'saturate':False}
             else:
                 if saturate:
                     # Do not staturate if struct field length is equal bitlen
```

Generating:
```
uint32_t dsdl_BoolRequest_encode_internal(dsdl_BoolRequest* source,
  void* msg_buf,
  uint32_t offset,
  uint8_t CANARD_MAYBE_UNUSED(root_item))
{
    canardEncodeScalar(msg_buf, offset, 1, (void*)&source->in); // 1
    offset += 1;

    return offset;
}
```

pavel-kirienko, 2019-02-02:
I am not sure whether it is safe to always truncate a boolean. In C, one would expect that upon casting to a boolean, 2 becomes True, which holds for saturation; if the value is truncated, however, you end up with 2 & 1 = False. I think it needs to be investigated instead why there is zero in `CANARD_INTERNAL_SATURATE_UNSIGNED(source->in, 0)`. Unless I am misunderstanding something.

spiremike, 2019-02-02:
You're correct about the truncation.

I think the reason the zero is is due to this part:
```
-                    'max_size':get_max_size(t.bitlen, False),
+                    'max_size':get_max_size(t.bitlen, True),
```

The call to `get_max_size` :
```
def get_max_size(bits, unsigned):
    if unsigned:
        return (2 ** bits) -1
    else:
        return (2 ** (bits-1)) -1
```

False treats the value as signed and the calculation gives a max size of 0, setting it to True it is treated as unsigned and correctly returns `1`. This introduces another issue though:
```
test/dsdl_Bool.c: In function 'dsdl_dsdl_BoolResponse_encode_internal':
test/dsdl_Bool.c:16:58: error: comparison of constant '1' with boolean expression is always false [-Werror=bool-compare]
 #define CANARD_INTERNAL_SATURATE_UNSIGNED(x, max) ( ((x) > max) ? max : (x) );
                                                          ^
test/dsdl_Bool.c:163:23: note: in expansion of macro 'CANARD_INTERNAL_SATURATE_UNSIGNED'
     source->success = CANARD_INTERNAL_SATURATE_UNSIGNED(source->success, 1)
                       ^
cc1: all warnings being treated as errors
```

I'm looking at another way to fix this portably just now

spiremike, 2019-02-02:
Changing the definition of `CANARD_INTERNAL_SATURATE_UNSIGNED` to use `>=` instead of `>` fixes the compiler warning and is functionally equivalent.

The following patch has this change and the change to calculate bool max size as an unsigned:

```
diff --git a/dsdl_compiler/libcanard_dsdl_compiler/__init__.py b/dsdl_compiler/libcanard_dsdl_compiler/__init__.py
index 9b66128..f441171 100644
--- a/dsdl_compiler/libcanard_dsdl_compiler/__init__.py
+++ b/dsdl_compiler/libcanard_dsdl_compiler/__init__.py
@@ -217,7 +217,7 @@ def type_to_c_type(t):
                     'post_cpp_type':'',
                     'cpp_type_comment':'bit len %d' % (t.bitlen, ),
                     'bitlen':t.bitlen,
-                    'max_size':get_max_size(t.bitlen, False),
+                    'max_size':get_max_size(t.bitlen, True),
                     'signedness':signedness,
                     'saturate':saturate}
             else:
diff --git a/dsdl_compiler/libcanard_dsdl_compiler/code_type_template.tmpl b/dsdl_compiler/libcanard_dsdl_compiler/code_type_template.tmpl
index b917f3d..9810f0e 100644
--- a/dsdl_compiler/libcanard_dsdl_compiler/code_type_template.tmpl
+++ b/dsdl_compiler/libcanard_dsdl_compiler/code_type_template.tmpl
@@ -15,7 +15,7 @@
 #endif

 #ifndef CANARD_INTERNAL_SATURATE_UNSIGNED
-#define CANARD_INTERNAL_SATURATE_UNSIGNED(x, max) ( ((x) > max) ? max : (x) );
+#define CANARD_INTERNAL_SATURATE_UNSIGNED(x, max) ( ((x) >= max) ? max : (x) );
 #endif

 #if defined(__GNUC__)
```

pavel-kirienko, 2019-02-02:
Very good. May I ask you to submit a pull request, please? Thanks!