OpenCyphal-Garage/libcyphal #209: Can not use cmake debug build with libuavcan

chemicstry, 2019-03-12:
I'm using libuavcan as a cmake module by including `libuavcan/CMakeLists.txt` and I can not build my project in debug without triggering libuavcan to be built in debug as well. This is because libuavcan sets `DEBUG_BUILD` based on cmake build type in here https://github.com/UAVCAN/libuavcan/blob/master/libuavcan/CMakeLists.txt#L16. Can we get an option to  set DEBUG_BUILD manually?

This wouldn't be much of a problem on its own, however, debug build fails without `UAVCAN_TOSTRING`:
```
[build] In file included from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/transport/dispatcher.hpp:14,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/node/scheduler.hpp:10,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/node/abstract_node.hpp:10,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/node/generic_publisher.hpp:9,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/src/node/uc_generic_publisher.cpp:5:
[build] C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/transport/outgoing_transfer_registry.hpp: In member function 'bool uavcan::OutgoingTransferRegistry::DeadlineExpiredPredicate::operator()(const uavcan::OutgoingTransferRegistryKey&, const uavcan::OutgoingTransferRegistry::Value&) const':
[build] C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/transport/outgoing_transfer_registry.hpp:89:34: error: 'const class uavcan::OutgoingTransferRegistryKey' has no member named 'toString'
[build]                               key.toString().c_str(), int(value.tid.get()));
[build]                                   ^~~~~~~~
[build] In file included from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/transport/dispatcher.hpp:14,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/node/scheduler.hpp:10,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/node/abstract_node.hpp:10,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/node/generic_subscriber.hpp:9,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/src/node/uc_generic_subscriber.cpp:5:
[build] C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/transport/outgoing_transfer_registry.hpp: In member function 'bool uavcan::OutgoingTransferRegistry::DeadlineExpiredPredicate::operator()(const uavcan::OutgoingTransferRegistryKey&, const uavcan::OutgoingTransferRegistry::Value&) const':
[build] C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/transport/outgoing_transfer_registry.hpp:89:34: error: 'const class uavcan::OutgoingTransferRegistryKey' has no member named 'toString'
[build]                               key.toString().c_str(), int(value.tid.get()));
[build]                                   ^~~~~~~~
```

And also another build error caused by a sign conversion warning:
```
[build] In file included from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/error.hpp:8,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/node/generic_publisher.hpp:8,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/src/node/uc_generic_publisher.cpp:5:
[build] C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/build_config.hpp: In static member function 'static void uavcan::IsDynamicallyAllocatable<T>::check()':
[build] C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/build_config.hpp:246:60: error: unsigned conversion from 'int' to 'unsigned int' changes the value of '-1' [-Werror=sign-conversion]
[build]          char dummy[(sizeof(T) <= MemPoolBlockSize) ? 1 : -1] = { '0' };
[build]                                                             ^
[build] In file included from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/error.hpp:8,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/node/generic_subscriber.hpp:8,
[build]                  from C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/src/node/uc_generic_subscriber.cpp:5:
[build] C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/build_config.hpp: In static member function 'static void uavcan::IsDynamicallyAllocatable<T>::check()':
[build] C:/Users/User/Documents/Dev/ChibiOSTests/AUXModule_cmake/deps/libuavcan/libuavcan/include/uavcan/build_config.hpp:246:60: error: unsigned conversion from 'int' to 'unsigned int' changes the value of '-1' [-Werror=sign-conversion]
[build]          char dummy[(sizeof(T) <= MemPoolBlockSize) ? 1 : -1] = { '0' };
```

Anyway, thank you for this great project!

thirtytwobits, 2019-05-04:
Sorry for the late response here. I've been busy with the v1 branch.

I'd rather make sure your debug build is working. If you want to build debug you should be able to. I'm able to build using the official Vagrant image that uses gcc5 (yeah, I know. That's pretty ancient). What compiler are you trying to use?

pavel-kirienko, 2019-05-04:
> I'd rather make sure your debug build is working. If you want to build debug you should be able to.

Not on embedded systems. A debug build of libuavcan v1 requires stuff which is unlikely to be available on an embedded system, such as `std::string`. This is a very questionable design and it would be great to avoid such huge build setting variability in the blue sky (no pressure though).

sprenger120, 2021-09-20:
If someone still has this issue and is including uavcan via subdirectory, you can do something like this 

```
set(PreviousBuildType "${CMAKE_BUILD_TYPE}")  
set(CMAKE_BUILD_TYPE  "Release")  
add_subdirectory(bus-node-base/libuavcan/libuavcan)  
set(CMAKE_BUILD_TYPE  "${PreviousBuildType}")
``` 

Let me compile my project in release and debug without the compile errors