OpenCyphal-Garage/libcyphal #211: building libuavcan on atollic ide for stm32

teklike, 2019-04-24:
I cloned the libuavcan library as a submodule into an stm32L4xxx project (C++ project) in Atollic TrueSTUDIO  IDE.  Invoking the builder, I get error:  Timestamp.hpp 
From Pavel's comments, I gather the DSDL compiler is not working?  What am I missing?

pavel-kirienko, 2019-04-24:
Let's see how you are invoking the DSDL compiler.

teklike, 2019-04-24:
Looking at the build log, I know the DSDL compiler is not being invoked. I am at a loss really as to how to invoke the DSDL compiler. I have tried to hack (which I thought would later help me write perhaps a pre-build-step), but not getting anywhere. So some guidance would be greatly appreciated

I thought, if I could, from the command line  run the python script that calls the compiler I would be able to generate the DSDL compiler output directories/files. Then  "manually" load these in the project. 
Later I can automate this.... but I seem not be able to do this. I have a big drawback that my IDE is running on windows10, so although it should not make much of a difference.... the way library is designed.... but nonetheless I feel I need to mention this. 

It might be a little out of the norm, but I can setup an online meeting (webex) at a time that is convenient to you, I can show you the environment and after the problem resolution I can document in detail the steps (this could be very helpful to a lot of folks, running their IDEs on windows and very eager to use the libuavcan. 

Below are some snapshots of the project file structure I am working with: 
1) first is the "project explorer" view showing where the sub module is located and its main folders/files:
[image]

2) Some unsuccessful guesses at what might run the compiler are below: 
c:\Projects_2\bms_021\Src\libuavcan\libuavcan\dsdl_compiler\build>cd lib

c:\Projects_2\bms_021\Src\libuavcan\libuavcan\dsdl_compiler\build\lib>cd libuavcan_dsdl_compiler

c:\Projects_2\bms_021\Src\libuavcan\libuavcan\dsdl_compiler\build\lib\libuavcan_dsdl_compiler>dir
 Volume in drive C has no label.
 Volume Serial Number is D677-811B

 Directory of c:\Projects_2\bms_021\Src\libuavcan\libuavcan\dsdl_compiler\build\lib\libuavcan_dsdl_compiler

04/24/2019  03:00 PM    <DIR>          .
04/24/2019  03:00 PM    <DIR>          ..
04/24/2019  02:17 PM            17,111 data_type_template.tmpl
04/24/2019  02:17 PM            48,857 pyratemp.py
04/24/2019  02:17 PM            12,402 __init__.py
               3 File(s)         78,370 bytes
               2 Dir(s)  185,984,700,416 bytes free

c:\Projects_2\bms_021\Src\libuavcan\libuavcan\dsdl_compiler\build\lib\libuavcan_dsdl_compiler>python __init__.py
Traceback (most recent call last):
  File "__init__.py", line 17, in <module>
    from .pyratemp import Template
ValueError: Attempted relative import in non-package

teklike, 2019-04-25:
it would be good to have a separate build step, that can be run just to invoke the DSDL to compile the structures placed in the same or planned heirarchy to transparently compile and generate the header files in C++. Now that  everything is in C++, it would be easy to run these in a C++ compiler on nearly all platform that provide a compliant C++ compiler.

pavel-kirienko, 2019-04-28:
The executable of the DSDL compiler is named `libuavcan_dsdlc`, and this is where you can find it in your cloned libuavcan repo: https://github.com/UAVCAN/libuavcan/blob/master/libuavcan/dsdl_compiler/libuavcan_dsdlc. This is how you invoke it:

```sh
$ libuavcan_dsdlc path/to/dsdl/uavcan
```

The usage is explained here https://github.com/UAVCAN/libuavcan_stm32 (see the README) and https://uavcan.org/Implementations/Libuavcan/Tutorials/8._Custom_data_types/.

The DSDL compiler is trivial to invoke, just run the script. If it doesn't work, please post detailed error messages here.

closed 2019-05-08 by pavel-kirienko

teklike, 2019-05-08:
beslam@swift-wks736 MINGW64 /c/Projects/Dsdl_TestingFolder/libuavcan/libuavcan/dsdl_compiler (master)
$ python libuavcan_dsdlc  /c/Projects/Dsdl_TestingFolder/libuavcan/dsdl
Internal error
Traceback (most recent call last):
  File "C:\Projects\Dsdl_TestingFolder\libuavcan\libuavcan\dsdl_compiler\pyuavcan\uavcan\dsdl\parser.py", line 606, in parse
    return self.parse_source(filename, source_text)
  File "C:\Projects\Dsdl_TestingFolder\libuavcan\libuavcan\dsdl_compiler\pyuavcan\uavcan\dsdl\parser.py", line 598, in parse_source
    raise ex
DsdlException: C:\Projects\Dsdl_TestingFolder\libuavcan\dsdl\uavcan\equipment\ahrs\1000.Solution.uavcan:5: Unknown namespace [uavcan]
Compiler failure
Traceback (most recent call last):
  File "libuavcan_dsdlc", line 61, in <module>
    dsdlc_run(args.source_dir, args.incdir, args.outdir)
  File "C:\Projects\Dsdl_TestingFolder\libuavcan\libuavcan\dsdl_compiler\libuavcan_dsdl_compiler\__init__.py", line 58, in run
    types = run_parser(source_dirs, include_dirs + source_dirs)
  File "C:\Projects\Dsdl_TestingFolder\libuavcan\libuavcan\dsdl_compiler\libuavcan_dsdl_compiler\__init__.py", line 97, in run_parser
    die(ex)
  File "C:\Projects\Dsdl_TestingFolder\libuavcan\libuavcan\dsdl_compiler\libuavcan_dsdl_compiler\__init__.py", line 90, in die
    raise DsdlCompilerException(str(text))
DsdlCompilerException: C:\Projects\Dsdl_TestingFolder\libuavcan\dsdl\uavcan\equipment\ahrs\1000.Solution.uavcan: Internal error: C:\Projects\Dsdl_Test
ingFolder\libuavcan\dsdl\uavcan\equipment\ahrs\1000.Solution.uavcan:5: Unknown namespace [uavcan]
C:\Projects\Dsdl_TestingFolder\libuavcan\dsdl\uavcan\equipment\ahrs\1000.Solution.uavcan: Internal error: C:\Projects\Dsdl_TestingFolder\libuavcan\dsd
l\uavcan\equipment\ahrs\1000.Solution.uavcan:5: Unknown namespace [uavcan]

pavel-kirienko, 2019-05-08:
The compiler expects a path to the root namespace, so the correct way in your case would be:

```
python libuavcan_dsdlc /c/Projects/Dsdl_TestingFolder/libuavcan/dsdl/uavcan
```
...instead of (wrong):
```
python libuavcan_dsdlc /c/Projects/Dsdl_TestingFolder/libuavcan/dsdl
```

teklike, 2019-05-08:
OK , let me try that

teklike, 2019-05-08:
yes, the command was executed! Thanks!!!
I am going to quickly check to see if Timestamps.hpp file was generated .... while you are still looking at the comments:)

teklike, 2019-05-08:
yes that solves the problem!
Thanks for the help. We can close this issue with the issue resolved