OpenCyphal-Garage/libcyphal #217: invalid namespace error in building custom Data, from DSDL compiler

teklike, 2019-05-16:
Running the libuavcan_dsdlc gives  invalid namespace error. I am placing a single data definition file in directory that is at the same level as libuavcan library. 
My understanding is that this should be accepted as a namespace by the DSDL compiler but according to the the Gitbash terminal listing below that does not seem to happen:
 beslam@swift-wks736 MINGW64 /c/Projects_2/RevdA6vg/Src (BmsLibuavcan)
$ python libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc  BmsCanMsg  -I/libuavcan/dsdl/uavcan
Internal error
Traceback (most recent call last):
  File "C:\Projects_2\RevdA6vg\Src\libuavcan\libuavcan\dsdl_compiler\pyuavcan\uavcan\dsdl\parser.py", line 606, in parse
    return self.parse_source(filename, source_text)
  File "C:\Projects_2\RevdA6vg\Src\libuavcan\libuavcan\dsdl_compiler\pyuavcan\uavcan\dsdl\parser.py", line 598, in parse_source
    raise ex
DsdlException: BmsCanMsg\BattVolt.uavcan: Invalid namespace name [BmsCanMsg]
Compiler failure
Traceback (most recent call last):
  File "libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc", line 61, in <module>
    dsdlc_run(args.source_dir, args.incdir, args.outdir)
  File "C:\Projects_2\RevdA6vg\Src\libuavcan\libuavcan\dsdl_compiler\libuavcan_dsdl_compiler\__init__.py", line 58, in run
    types = run_parser(source_dirs, include_dirs + source_dirs)
  File "C:\Projects_2\RevdA6vg\Src\libuavcan\libuavcan\dsdl_compiler\libuavcan_dsdl_compiler\__init__.py", line 97, in run_parser
    die(ex)
  File "C:\Projects_2\RevdA6vg\Src\libuavcan\libuavcan\dsdl_compiler\libuavcan_dsdl_compiler\__init__.py", line 90, in die
    raise DsdlCompilerException(str(text))
DsdlCompilerException: BmsCanMsg\BattVolt.uavcan: Internal error: BmsCanMsg\BattVolt.uavcan: Invalid namespace name [BmsCanMsg]
BmsCanMsg\BattVolt.uavcan: Internal error: BmsCanMsg\BattVolt.uavcan: Invalid namespace name [BmsCanMsg]

My diectory structure is as below:
[image]

I seem to be missing something....

thirtytwobits, 2019-05-17:
I can try to help (@pavel-kirienko is the v0 expert). Can you provide your BattVolt.uavcan file so I can try to reproduce?

pavel-kirienko, 2019-05-17:
The error is because the name of the namespace is not lowercase. (Our new library PyDSDL, on the other hand, does not enforce rules that are not mandatory.)

closed 2019-05-17 by pavel-kirienko

teklike, 2019-05-17:
yes, that was the problem. Thanks guys!