OpenCyphal-Garage/libcyphal #226: integrate clang-tidy with v1 build

thirtytwobits, 2019-05-26:
run clang-tidy at build-time.

pavel-kirienko, 2024-04-29:
Clang-Tidy should be integrated via `CXX_CLANG_TIDY` and Sonar needs to be invoked on pull request builds or when the commit is tagged with `#sonar`.

@serges147 I would like to prioritize this because the way the codebase is now we're already going to have to go back and modify it to make Clang-Tidy and Sonar accept it.

serges147, 2024-04-29:
@pavel-kirienko Pavel, I'm right that I can use exactly the same approach as in CETL in terms of Sonar? Also where to get `clang-tidy` config file?

pavel-kirienko, 2024-04-29:
1. Yes; alternatively, you could use Dyshlo as a reference.
2. Let's start with this one (it may require some tweaking but we'll get to that later):

```yaml
Checks: >-
  boost-*,
  bugprone-*,
  cert-*,
  clang-analyzer-*,
  cppcoreguidelines-*,
  google-*,
  hicpp-*,
  llvm-*,
  misc-*,
  modernize-*,
  performance-*,
  portability-*,
  readability-*,
  -cppcoreguidelines-avoid-const-or-ref-data-members,
  -google-readability-todo,
  -readability-avoid-const-params-in-decls,
  -readability-identifier-length,
  -llvm-header-guard,
  -llvm-include-order,
  -*-use-trailing-return-type,
  -*-named-parameter,
CheckOptions:
  - key:   readability-function-cognitive-complexity.Threshold
    value: '90'
  - key:   readability-magic-numbers.IgnoredIntegerValues
    value: '1;2;3;4;5;8;10;16;20;32;60;64;100;128;256;500;512;1000'
WarningsAsErrors: '*'
HeaderFilterRegex: '.*\.hpp'
AnalyzeTemporaryDtors: false
FormatStyle: file
```

closed 2024-12-05 by pavel-kirienko