OpenCyphal-Garage/libcyphal #228: Consider allowing c++14 by default

thirtytwobits, 2019-05-28:
@antoinealb has requested that the logic in `build_config.hpp` be modified to allow compiling for C++14 without an explicit `-DUAVCAN_CPP_VERSION` override. 

## For
@antoinealb feels that the delta between C++11 and C++14 is trivial enough that we should allow C++14 to be used without an override and can safeguard against C++14 dependencies sneaking into the library by building our CI using C++11.

## Against
@pavel-kirienko and @thirtytwobits are reticent given the lack of testing against C++14 targets and that we explicitly do not want to require C++14. We've settled on C++11 as our baseline for v0 and do not expect this to change.

pavel-kirienko, 2019-05-28:
I think we should start testing against C++14 in the CI and add the support @antoinealb is asking for.

antoinealb, 2019-05-28:
I can provide feedback on C++14 compatibility once I finish re-writing our build system. However I cannot commit to providing comprehensive and long-term support or testing for it, and I understand if the core team does not want to officially support it .

One other option that I see as perhaps more aligned with the interest of the maintainers would be to add a note in the README explaining that C++{14, 17, 20} is not officially supported but that a define allows it to be used, perhaps with a link to the relevant issues.

pavel-kirienko, 2019-06-04:
> I understand if the core team does not want to officially support it .

I am not sure we're all aligned on this one. I see no problem with modifying the v0 testing suite to make it run tests against C++14 and C++17. There should be no issues since we actually use libuavcan in a C++17 embedded application successfully.

closed 2024-04-04 by pavel-kirienko