OpenCyphal-Garage/libcyphal #234: Build Nunavut templates for v1 types.

thirtytwobits, 2019-05-29:
This is a placeholder. Hopefully I can break this down in to a few tasks rather that one big uber-task.

pavel-kirienko, 2019-06-04:
Relevant notes on near-zero-cost serialization with the help of bit length analysis available in PyDSDL v1.0: https://github.com/UAVCAN/pydsdl/pull/24

Relevant excerpts from the PyUAVCAN (de)serialization code generation:
- https://github.com/UAVCAN/pyuavcan/blob/d986161a5775af79e7367f16da0616c198838775/pyuavcan/dsdl/_templates/serialization.j2#L55-L127
- https://github.com/UAVCAN/pyuavcan/blob/d986161a5775af79e7367f16da0616c198838775/pyuavcan/dsdl/_templates/deserialization.j2#L27-L126

thirtytwobits, 2020-02-18:
Closing as this is superseded by https://github.com/UAVCAN/nunavut/issues/91.

closed 2020-02-18 by thirtytwobits