OpenCyphal-Garage/libcyphal #254: Infinite loop in the AVL tree

pavel-kirienko, 2019-06-24:
It has been reported that there is allegedly an infinite loop in the AVL tree:

https://forum.uavcan.org/t/libuavcan-cubemx-freertos-systemworkbenchforstm32/430/9?u=pavel.kirienko

dpslwk, 2019-07-16:
fix suggested by kolos450 in the linked thread got this working for me on an STM32L433 under FreeRTOS

pavel-kirienko, 2019-07-30:
For reference, this is the suggested fix copy-pasted from the forum verbatim:

> Hi. I've got the same issue. There is a typo in the appendToEndOf method:
> ```
> target = head->equal_keys;
> ```
> should be 
> ```
> target = target->equal_keys;
> ```
> The case happens when the driver is busy and all the frames produced from a single message are sent to a linked list.

@thirtytwobits should we just go ahead and fix this, seeing as @Zarkopafilis is unavailable?

pavel-kirienko, 2019-07-30:
Related: https://forum.uavcan.org/t/how-to-support-getnodeinfo-and-getdatatypeinfo-on-stm32/577

thirtytwobits, 2019-07-30:
I'll take a look

thirtytwobits, 2019-08-05:
All fixed. Thanks for the bug report and sorry for the slow response.

closed 2019-08-05 by thirtytwobits