OpenCyphal-Garage/libcyphal #256: IRQ for STM32L4

teklike, 2019-07-15:
I am using libuavcan to send single frame broadcast messages.  On going through the debugging process, I noticed in the uc_stm32_can.cpp file,  there are defines for STM32F3XX and STM32 processors running NUTTX etc. with specific mappings for CAN interrupt Handlers.  But although there is mention of freeRTOS on STM32 .... there is no mapping. Is this truly correct or we need to add to this file similar mappings for STM32L4 running FREERTOS? (And if so.... could  I be pointed to a location, where there is an example for this?) 

My exact processor is STM32L4A6VG, running FreeRTOS.

teklike, 2019-07-15:
running the file as is, I get the following errors:
[image]

teklike, 2019-07-16:
[stm32l4a6xx.zip](https://github.com/UAVCAN/libuavcan/files/3394696/stm32l4a6xx.zip)
[stm32l4xx_hal_conf.zip](https://github.com/UAVCAN/libuavcan/files/3394699/stm32l4xx_hal_conf.zip)

Processor register and config Files for stm32l4a6xx

dpslwk, 2019-07-16:
https://github.com/UAVCAN/libuavcan_stm32/pull/3

pavel-kirienko, 2019-07-30:
My best suggestion is to abandon the stock STM32 driver we have and use your own. You can use the stock one as a starting point (MIT license); drop all preprocessor branches your platform does not require (that'll cut the code size by a third I expect) and then add the missing bits in there.

I think in v1 we might need to have a narrower focus for this driver. Maybe there should be just one implementation for baremetal and a dedicated implementation per RTOS. If you feel strongly about this, consider jumping onto our [dev call tomorrow at 18:00 UTC](https://forum.uavcan.org/t/weekly-dev-call/173].

closed 2019-07-30 by pavel-kirienko