OpenCyphal-Garage/libcyphal #258: CMake ERROR: ModuleNotFoundError: No module named 'uavcan'

jiapei100, 2019-07-24:
Is it possible to build **C/C++ ONLY, without Python** ?

```
[  1%] Running dsdl compiler
cd ....../UAVCAN/libuavcan/libuavcan && /usr/bin/python ....../UAVCAN/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc test/dsdl_test/root_ns_a test/dsdl_test/root_ns_b ....../UAVCAN/libuavcan/libuavcan/../dsdl/uavcan -Oinclude/dsdlc_generated
Traceback (most recent call last):
  File "....../UAVCAN/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc", line 59, in <module>
    from libuavcan_dsdl_compiler import run as dsdlc_run
  File "....../UAVCAN/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 18, in <module>
    from uavcan import dsdl
ModuleNotFoundError: No module named 'uavcan'
```

thirtytwobits, 2019-07-26:
Unfortunately, no. The python is used to generate C++ types from DSDL. These types are part of the core protocol. We will improve this integration with UAVCAN v1 but regardless we will always require python generation from DSDL as part of the libuavcan build.

thirtytwobits, 2019-07-26:
As for your issue, don't forget to `git submodule update --init`

closed 2019-07-26 by thirtytwobits

jiapei100, 2019-07-26:
@thirtytwobits You are right... Problem solved...

deworben, 2022-08-30:
I had to run this with force to get this to work: `git submodule update --init --recursive --force`