OpenCyphal-Garage/libcyphal #271: stm32f107rct6 comm issue

Tal-seven, 2019-12-31:
I have 4 boards configured for uavcan, 2 stm32f4 custom pcb's, one imx6 soc board and an stm32f107rct6 custom pcb. The stm32f4 nodes and imx6 node work flawlessly but the stm32f107 node when configured with uavcan does not communicate nor does it output any sensible traffic on can bus. 

In my configuration i am not using Chibios HAL, I am using standard peripheral library that is responsible for initializing and configuring the clocks and gpio. The reason for not using Chibios HAL is that i have another HAL library that i've been using for a long time and cant switch to any other HAL for now. Following is the hardware initialization code i use:

`void CAN_Config(void)
{
  GPIO_InitTypeDef  GPIO_InitStructure;
	CAN_InitTypeDef CAN_InitStructure;
  RCC_APB2PeriphClockCmd(RCC_APB2Periph_AFIO | RCC_APB2Periph_GPIOB, ENABLE);
     
  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_IPU;
  GPIO_InitStructure.GPIO_Pin = GPIO_Pin_8;
  GPIO_Init(GPIOB, &GPIO_InitStructure);

	GPIO_InitStructure.GPIO_Pin = GPIO_Pin_12;
	GPIO_Init(GPIOB,&GPIO_InitStructure);
	

  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_AF_PP;
  GPIO_InitStructure.GPIO_Speed = GPIO_Speed_50MHz;
  GPIO_InitStructure.GPIO_Pin = GPIO_Pin_9;
  GPIO_Init(GPIOB, &GPIO_InitStructure);

	GPIO_InitStructure.GPIO_Pin = GPIO_Pin_13;
	GPIO_Init(GPIOB,&GPIO_InitStructure);

	GPIO_PinRemapConfig(GPIO_Remap1_CAN1,ENABLE);
  RCC_APB1PeriphClockCmd(RCC_APB1Periph_CAN1 | RCC_APB1Periph_CAN2, ENABLE);  
  CAN_DeInit(CAN1);
  CAN_DeInit(CAN2);

  CAN_StructInit(&CAN_InitStructure);

  CAN_InitStructure.CAN_TTCM = DISABLE;
  CAN_InitStructure.CAN_ABOM = DISABLE;
  CAN_InitStructure.CAN_AWUM = DISABLE;
  CAN_InitStructure.CAN_NART = DISABLE;
  CAN_InitStructure.CAN_RFLM = DISABLE;
  CAN_InitStructure.CAN_TXFP = ENABLE;
  CAN_InitStructure.CAN_Mode = CAN_Mode_Normal;
  CAN_InitStructure.CAN_SJW = CAN_SJW_1tq;  
  CAN_InitStructure.CAN_BS1 = CAN_BS1_3tq;
  CAN_InitStructure.CAN_BS2 = CAN_BS2_2tq;
 
  CAN_InitStructure.CAN_Prescaler = 12;
 CAN_Init(CAN1,&CAN_InitStructure);
  CAN_Init(CAN2, &CAN_InitStructure);
}`

Its a straight forward initialization. Transmit receive using std peripheral library after this works fine but after CanIface::init is used to initialize the node and uavcan::Iface. The output looks like that of a node which does not receive an ack and keeps retransmitting.

Can dump fails to sniff any packets in this state and uavcan_monitor does not recognize the node as well.  

Following is the initialization code for the node:

```
`void hwInit()	{
	int res=0;
	uint32_t bitrate =500000;
	do{
		//res = can_.init([](){hw::Timer.udelay(can_.getRecommendedListeningDelay().toUSec());},bitrate);
		res = can_.init(bitrate);
		if(res >= 0)
		{
			sys::debug(Debug_Level::Info,"Init can successful\r\n");
		}
		else
		{
			sys::debug(Debug_Level::Info,"Init can failed\r\n");
		}
	} while (res < 0);
}
class batteryNode : public chibios_rt::BaseStaticThread<8192> {

private:
	void startNode(Node& node) {
		node.setName("uavcan.BQ34Z100");
		node.setNodeID(30);

		while(true) {
			if(int node_res = node.start();node_res < 0)
			{
				sys::debug(sys::Debug_Level::Error, "Failed to start Battery Node\r\n");
			}
			else
			{
				sys::std_out("Started Node %d\r\n",node.getNodeID().get());
				break;
			}
		}
	}

public:
	static chibios_rt::ThreadReference batteryThd;
	void main()	{
		Node& node = getNode();
		hwInit();
		startNode(node);
		stm32plus::I2C::Parameters params;
		BQ34Z100 gauge(params);

		if(!gauge.initialise())
			sys::debug(sys::Debug_Level::Error,"Failed to Initialise BatteryGauge\r\n");
		else
			sys::debug(sys::Debug_Level::Info,"Battery Gauge Initialised\r\n");

		gauge.unsealDataFlash();
		gauge.switchFullAccess();
//		gauge.configureGauge();

		uavcan::Publisher<uavcan::equipment::power::Battery> bat_pub(node);
		if(const int bat_res = bat_pub.init();bat_res < 0)
		{
			sys::debug(sys::Debug_Level::Error,"Failed to start battery publisher\r\n");
			batteryThd.stop();
		}
		
		uavcan::equipment::power::Battery batmsg;
		node.setModeOperational();
		node.setHealthOk();
		
		while(true)	{
			if(int spinres=node.spin(uavcan::MonotonicDuration::fromMSec(100));spinres < 0)
			{
				sys::debug(sys::Debug_Level::Warn,"batteryThread spin error\r\n");
			}

			bat_pub.broadcast(batmsg);
			print_nodeInfo();
		}
	}
	
}bq34_node;
`
```
 
Please do let me know on how to diagnose the issue further. This is my cmake configuration for libuavcan
```
`if(LIBUAVCAN_SUPPORT)
	add_definitions(
				-DUAVCAN_STM32_TIMER_NUMBER=5
			        -DUAVCAN_STM32_NUM_IFACES=2
				-DUAVCAN_STM32_CHIBIOS=1
				-DUAVCAN_CPP_VERSION=UAVCAN_CPP11
				-DUAVCAN_STM32_IRQ_PRIORITY_MASK=4
				 )
find_library(uavcanlib 
	            NAMES uavcan libuavcan  
		   HINTS "${CMAKE_BINARY_DIR}/thirdparty/libuavcan/libuavcan")

find_library(uavcanlibstm32 
		   NAMES uavcan_stm32_driver libuavcan_stm32_driver 
		HINTS "${CMAKE_BINARY_DIR}/thirdparty/libuavcan/libuavcan_drivers/stm32/driver/")

endif(LIBUAVCAN_SUPPORT)`
```

pavel-kirienko, 2019-12-31:
> Its a straight forward initialization. Transmit receive using std peripheral library after this works fine but after CanIface::init is used to initialize the node and uavcan::Iface. The output looks like that of a node which does not receive an ack and keeps retransmitting.

If I understood you correctly, things break after you invoke `uavcan::CanIface::init()`. The most common cause for issues here is an incorrect bit rate configuration, which in turn could be caused by the bit timing derivation method making an incorrect assumption about the clock rate of the bxCAN  macrocell:

https://github.com/UAVCAN/libuavcan_stm32/blob/4e0a24756090d218b469f561d7e0d35e5f2f5427/driver/src/uc_stm32_can.cpp#L205-L215

You can check this by probing `CAN_TX` with a logic analyzer.

The root of the problem, however, is not there. Rather, your problem is that you are essentially using two different drivers to do the same job concurrently: first, you initialize the hardware using SPL, then you do the same thing again using the stock driver shipped with Libuavcan. You should either discard SPL or make your own driver (it could be based on the stock one) to eliminate the redundancy.

P.S. it's best to ask for assistance [on the forum](https://forum.uavcan.org/) rather than here. This place is mostly intended for bug reports and feature requests.

Tal-seven, 2020-01-01:
Thank you for the response @pavel-kirienko. The issue was with the wrong configuration of "pclk".

closed 2020-01-01 by Tal-seven