OpenCyphal-Garage/libcyphal #276: Consider renaming the library

pavel-kirienko, 2020-02-05:
Libuavcan implements the UAVCAN/CAN transport only. When UAVCAN/UDP or some other transport is around, the name Libuavcan will be causing confusion. We should consider finding a better name for the library that reflects its CAN-focused nature better. A possible alternative is **libuavcancan** :dancer:.

LorenzMeier, 2020-02-05:
Given that there are limited resources to work on UAVCAN I would make sure you're not getting ahead of yourself. At a point where the core maintenance team is already overloaded implementing the new revision for CAN adding another physical layer seems a dilution of critical resources.

There is no great standard for CAN for drones, but there are dozens of realtime pub/sub protocols to choose from on a UDP transport, so the uptake there is much harder.

pavel-kirienko, 2020-02-05:
The cost of renaming is strongly correlated with adoption, so if we expect that it might become necessary in the future, it is rational to do it soon. It does not imply that we are already dedicating resources to other transports now.

LorenzMeier, 2020-02-05:
Cool, that's reassuring. Thinking ahead is great!

pavel-kirienko, 2020-03-06:
We decided to keep it as-is because we do intend to engineer the support for other transports into Libuavcan. CAN FD is obviously the number one target. Follow the details on the forum.

closed 2020-03-06 by pavel-kirienko

thirtytwobits, 2020-03-06:
The dream is dead.

[image]

(but seriously: this is the right decision)