OpenCyphal-Garage/libcyphal #278: Missing libuavcan_dsdlc in v1.0

jxltom, 2020-02-19:
I would like to try libuavan v1.0 as suggested in https://uavcan.org/, but I found the ```libuavcan_dsdlc``` tool is missing in https://github.com/UAVCAN/libuavcan/tree/uavcan-v1.0. Besides, it looks like the there is no docs and examples in v1.0, also can drivers are also missing.

What is the status of libuavcan v1.0? Should I still use v0 instead? Many thanks!

thirtytwobits, 2020-02-19:
I'm very sorry but libuavcan v1.0 isn't ready yet :( It's not abandoned though, it's just that the standardization process for the core specification is taking all my development time up right now. We are looking for additional contributors to speed things up if you are interested but I will be back to this within a month or two.

You can look at libcanard for the C implementation that is progressing more quickly and pyuavcan v1 is ready to go.

Also, libuavcan v1.0 will no longer using libuavcan_dsdlc. Instead we will use [nunavut](https://github.com/UAVCAN/nunavut) to generate C++ from DSDL in this version. Nunavut will become the unified code generator for the project when it is complete. You can watch my progress on C++ type serialization on my fork [here](https://github.com/thirtytwobits/nunavut/tree/feature/91). Sorry again. Please contribute, if not check back in a few months for more progress.

V0 is mature so using it is always an option. The V1 specification is in "alpha" and we are working with industry leaders to make it a formal specification with long term support.

jxltom, 2020-02-20:
Thanks for the explanation. I will keep using v0 for now and look forward to the release of v1!

pavel-kirienko, 2020-02-20:
Closing as duplicate of #277.

closed 2020-02-20 by pavel-kirienko