OpenCyphal-Garage/libcyphal #279: Tx packet timeout in socket can driver

mitchpi, 2020-02-20:
Hi Guys,

I'm doing a bit of testing of Uavcan on linux and I'm running into some issue with the driver interface.

Some packets are not looped back from the driver and are therefor not decremented from the frames_in_socket_tx_queue_ so the queue quickly fills and no more messages are sent. Still debugging why the loopback messages are not received. It only happens on broadcast messages and when there are no other nodes on the bus, suspect the messages are not being ACKd.

Why is there no timeout on messages in the frames_in_socket_tx_queue_? It appears there is one in the stm driver. Is this a feature that should be added? It they are not removed the uavcan stack has to be reset to continue sending messages.

pavel-kirienko, 2020-02-20:
There seems to be a problem with some SocketCAN drivers on GNU/Linux. Please read this thread: https://forum.uavcan.org/t/loopback-false-problems/338. A possible solution is to disable the TX queue scheduling as suggested there and just enqueue all outgoing frames immediately into SocketCAN. You can control the size of its own TX queue as `ifconfig slcan0 txqueuelen 100` as recommended in https://uavcan.org/Implementations/Libuavcan/Tutorials/2._Node_initialization_and_startup/#using-slcan-tunneling-can-over-serial-port.

> Why is there no timeout on messages in the frames_in_socket_tx_queue_?

Because on a properly functioning SocketCAN implementation every CAN frame sent into the queue will be confirmed if it is transmitted to the bus. As long as the frame is not confirmed, it is safe to assume that it is not transmitted, so there is no point in sending extra frames into the queue. Note that it may create a priority inversion problem but there are limitations in the SocketCAN API we can't do much about.

In v1, the interface is going to be built differently.

mitchpi, 2020-02-21:
Thanks for the link!!

Does that mean the issue is resolved in v1? I can see a bunch of pre-release versions, is there at ETA on a release version of that?

Do you know if the socketcan issue has been resolved in the latest kernel?

pavel-kirienko, 2020-02-21:
> Does that mean the issue is resolved in v1? I can see a bunch of pre-release versions, is there at ETA on a release version of that?

See https://github.com/UAVCAN/libuavcan/issues/278#issuecomment-588067582. If you or your teammates could lend us a hand, we could finish it up sooner.

> Do you know if the socketcan issue has been resolved in the latest kernel?

The kernel is fine. The problem is that some lower-layer SocketCAN drivers (that work under the SocketCAN core) are misbehaving. What hardware are you using? I think the easiest solution is to patch up the Libuavcan SocketCAN driver as described in that thread I linked earlier.

mitchpi, 2020-02-21:
We're currently using a zync microzed embedded linux board but will be updating to another embedded platform soon.

Are you waiting for a fix in the lower layer socketcan drivers or is the patch in libuavcan the way forward? We will need this fix so do you want a pull request for the changes? I was unable to run the standard demo tutorial code so it might be worth patching uavcan rather than waiting for a socketcan fix.

I will try to find some time over the coming sprints to help out.

pavel-kirienko, 2020-02-21:
> We will need this fix so do you want a pull request for the changes?

I think it's a rather common problem so yes, a pull request would be nice.

> I will try to find some time over the coming sprints to help out.

Fantastic.

mitchpi, 2020-02-26:
I've done a bit more investigating and it appears that socketcan is buffering the loopback packets somewhere or delaying the transmit of the packets until it has 7-8 packets. The loopbacks all seem to come through but the transmitted packet receives the loopback response 7-8 transmit packets later. Ie: the loopback for transmitted packet 1 comes through after transmitted packet 8 is sent, the loopback for transmitted packet 2 comes through after transmitted packet 9 is sent. 

The number of received packets from different nodes does not seem effect the loopback response. I have other can nodes sending packets and no matter how many packets are received inbetween the 8 transmits it still takes 7-8 for the loopback to come through.

I have also tried running multiple application instances on the same linux can device and if I have 2 app instances running the loopback response comes through after 4 transmits (ie the 8 packet buffer is split between the 2 instances), if I have 3 instances running it will split the 8 packet buffer between the app instances, some get 2 or 3 or 4, always maintaining the 8 packets until the response is received.

My node is configured as follows:
`[root@node1 ~]# ip -details -statistics link show can0
2: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 100
    link/can  promiscuity 0
    can state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 100
          bitrate 992063 sample-point 0.666
          tq 335 prop-seg 0 phase-seg1 1 phase-seg2 1 sjw 1
          xilinx_can: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..256 brp-inc 1
          clock 23809524
          re-started bus-errors arbit-lost error-warn error-pass bus-off
          0          9991835    0          3          9990558    2         numtxqueues 1 numrxqueues 1
    RX: bytes  packets  errors  dropped overrun mcast
    96274713   12414580 1295    6346    0       0
    TX: bytes  packets  errors  dropped carrier collsns
    14988      2131     9990167 7       0       0
`

So I don't think canuav requires any changes except to increase the max_frames_in_socket_tx_queue  to 10 or so in the socketcan.hpp file.

pavel-kirienko, 2020-02-26:
Okay. Would you be up to send a pull request with that change?

mitchpi, 2020-02-26:
Yes no problem

closed 2020-02-26 by mitchpi