OpenCyphal-Garage/libcyphal #312: fatal error: uavcan/Timestamp.hpp: No such file or directory

furkanmola, 2020-10-01:
Hi,
I'm trying to compile `px4_fmu-v3` using `make px4_fmu-v3`, but I'm getting the error:
`[1/1332] Performing build step for 'px4io_firmware'
[212/214] Linking CXX executable px4_io-v2_default.elf
Memory region         Used Size  Region Size  %age Used
           flash:       56520 B        60 KB     91.99%
            sram:        3832 B         8 KB     46.78%
[214/214] Creating /home/furkan/Firmwa...d/px4io_firmware/px4_io-v2_default.px4
[495/1332] Building CXX object src/dri...s/uavcan.dir/src/driver/uc_can.cpp.obj
FAILED: /usr/bin/ccache /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++  -DCONFIG_ARCH_BOARD_PX4_FMU_V3 -DUAVCAN_CPP_VERSION=UAVCAN_CPP03 -DUAVCAN_DRIVER=uavcan_stm32 -DUAVCAN_IMPLEMENT_PLACEMENT_NEW=1 -DUAVCAN_MEM_POOL_BLOCK_SIZE=48 -DUAVCAN_NO_ASSERTIONS -DUAVCAN_PLATFORM=generic -DUAVCAN_STM32_NUM_IFACES=2 -DUAVCAN_STM32_NUTTX=1 -DUAVCAN_STM32_TIMER_NUMBER=5 -D_SYS_CDEFS_H_ -D_SYS_REENT_H_ -D__CUSTOM_FILE_IO__ -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I../../boards/px4/fmu-v3/src -I../../platforms/nuttx/src/px4/common/include -I. -Isrc/lib -I../../platforms/nuttx/src/px4/stm/stm32f4/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -I../../src/drivers/uavcan/libuavcan/./libuavcan/include -I../../src/drivers/uavcan/libuavcan/./libuavcan/include/dsdlc_generated -I../../src/drivers/uavcan/libuavcan/libuavcan/include/dsdlc_generated -I../../src/drivers/uavcan/libuavcan/libuavcan/include -isystem ../../platforms/nuttx/NuttX/include/cxx -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Wall -Wextra -Wundef -std=c++11 -Os -DNDEBUG   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -nostdinc++ -fno-sized-deallocation -Wno-cast-align -Wno-deprecated-copy -Wno-address-of-packed-member -std=gnu++14 -MD -MT src/drivers/uavcan/libuavcan/libuavcan/CMakeFiles/uavcan.dir/src/driver/uc_can.cpp.obj -MF src/drivers/uavcan/libuavcan/libuavcan/CMakeFiles/uavcan.dir/src/driver/uc_can.cpp.obj.d -o src/drivers/uavcan/libuavcan/libuavcan/CMakeFiles/uavcan.dir/src/driver/uc_can.cpp.obj -c ../../src/drivers/uavcan/libuavcan/libuavcan/src/driver/uc_can.cpp
In file included from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/driver/system_clock.hpp:11,
                 from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/driver/can.hpp:12,
                 from ../../src/drivers/uavcan/libuavcan/libuavcan/src/driver/uc_can.cpp:6:
../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/time.hpp:12:10: fatal error: uavcan/Timestamp.hpp: No such file or directory
   12 | #include <uavcan/Timestamp.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[495/1332] Building CXX object src/dri...vcan.dir/src/node/uc_scheduler.cpp.obj
FAILED: /usr/bin/ccache /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++  -DCONFIG_ARCH_BOARD_PX4_FMU_V3 -DUAVCAN_CPP_VERSION=UAVCAN_CPP03 -DUAVCAN_DRIVER=uavcan_stm32 -DUAVCAN_IMPLEMENT_PLACEMENT_NEW=1 -DUAVCAN_MEM_POOL_BLOCK_SIZE=48 -DUAVCAN_NO_ASSERTIONS -DUAVCAN_PLATFORM=generic -DUAVCAN_STM32_NUM_IFACES=2 -DUAVCAN_STM32_NUTTX=1 -DUAVCAN_STM32_TIMER_NUMBER=5 -D_SYS_CDEFS_H_ -D_SYS_REENT_H_ -D__CUSTOM_FILE_IO__ -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I../../boards/px4/fmu-v3/src -I../../platforms/nuttx/src/px4/common/include -I. -Isrc/lib -I../../platforms/nuttx/src/px4/stm/stm32f4/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -I../../src/drivers/uavcan/libuavcan/./libuavcan/include -I../../src/drivers/uavcan/libuavcan/./libuavcan/include/dsdlc_generated -I../../src/drivers/uavcan/libuavcan/libuavcan/include/dsdlc_generated -I../../src/drivers/uavcan/libuavcan/libuavcan/include -isystem ../../platforms/nuttx/NuttX/include/cxx -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Wall -Wextra -Wundef -std=c++11 -Os -DNDEBUG   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -nostdinc++ -fno-sized-deallocation -Wno-cast-align -Wno-deprecated-copy -Wno-address-of-packed-member -std=gnu++14 -MD -MT src/drivers/uavcan/libuavcan/libuavcan/CMakeFiles/uavcan.dir/src/node/uc_scheduler.cpp.obj -MF src/drivers/uavcan/libuavcan/libuavcan/CMakeFiles/uavcan.dir/src/node/uc_scheduler.cpp.obj.d -o src/drivers/uavcan/libuavcan/libuavcan/CMakeFiles/uavcan.dir/src/node/uc_scheduler.cpp.obj -c ../../src/drivers/uavcan/libuavcan/libuavcan/src/node/uc_scheduler.cpp
In file included from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/driver/system_clock.hpp:11,
                 from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/driver/can.hpp:12,
                 from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/transport/can_io.hpp:17,
                 from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/transport/frame.hpp:10,
                 from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/transport/transfer_receiver.hpp:10,
                 from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/transport/transfer_listener.hpp:11,
                 from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/transport/dispatcher.hpp:13,
                 from ../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/node/scheduler.hpp:10,
                 from ../../src/drivers/uavcan/libuavcan/libuavcan/src/node/uc_scheduler.cpp:5:
../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/time.hpp:12:10: fatal error: uavcan/Timestamp.hpp: No such file or directory
   12 | #include <uavcan/Timestamp.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[495/1332] Building CXX object src/dri....dir/src/marshal/uc_bit_stream.cpp.obj
FAILED: /usr/bin/ccache /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++  -DCONFIG_ARCH_BOARD_PX4_FMU_V3 -DUAVCAN_CPP_VERSION=UAVCAN_CPP03 -DUAVCAN_DRIVER=uavcan_stm32 -DUAVCAN_IMPLEMENT_PLACEMENT_NEW=1 -DUAVCAN_MEM_POOL_BLOCK_SIZE=48 -DUAVCAN_NO_ASSERTIONS -DUAVCAN_PLATFORM=generic -DUAVCAN_STM32_NUM_IFACES=2 -DUAVCAN_STM32_NUTTX=1 -DUAVCAN_STM32_TIMER_NUMBER=5 -D_SYS_CDEFS_H_ -D_SYS_REENT_H_ -D__CUSTOM_FILE_IO__ -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I../../boards/px4/fmu-v3/src -I../../platforms/nuttx/src/px4/common/include -I. -Isrc/lib -I../../platforms/nuttx/src/px4/stm/stm32f4/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/c [... cut ...]
../../src/drivers/uavcan/libuavcan/./libuavcan/include/uavcan/time.hpp:12:10: fatal error: uavcan/Timestamp.hpp: No such file or directory
   12 | #include <uavcan/Timestamp.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[495/1332] Building CXX object src/dri....dir/src/marshal/uc_float_spec.cpp.obj
ninja: build stopped: subcommand failed.
make: *** [Makefile:226: px4_fmu-v3] Error 1
`
When I try to compile `px4_sitl` using `make px4_sitl`, I do not have any problem. Additionaly, I've tried 
`git submodule update --init --recursive --force`.

pavel-kirienko, 2020-10-01:
Did the FAQ not help? https://legacy.uavcan.org/Implementations/Libuavcan/FAQ/

furkanmola, 2020-10-01:
> Did the FAQ not help? https://legacy.uavcan.org/Implementations/Libuavcan/FAQ/

Unfortunately, no.

pavel-kirienko, 2020-10-01:
I can't tell why the DSDL compiler is not working, please post the build log in its entirety. Also, while doing so, please ensure that UAVCAN is enabled by default for px4_fmu-v3.

pavel-kirienko, 2020-10-02:
If this is still valid, please follow up on the forum. This issue is not really related to libuavcan, it is caused by your build system.

closed 2020-10-02 by pavel-kirienko